LaserWeb/LaserWeb4 #528: Project status and community?

msurguy, 2018-11-28:
Hello!

As we know, lots and lots of people are using LaserWeb and it is a great piece of software built over the years of hard work. 

I was wondering what is happening with LaserWeb as far as the project since there is no activity for a while and wanted to explore some options and understand about the future of LaserWeb.

Could the original authors of the project pitch in and tell about their vision for the project and maybe it's a good time to think about the next steps for it? 

Thank you!

FabriceSalvaire, 2018-12-11:
Good question ! Close issues like this one https://github.com/LaserWeb/LaserWeb4/issues/524 makes me doubtful.

cprezzi, 2018-12-31:
Pull requests are welcome and will be accepted if it make sense.

The AGPL license was choosen to make sure that the source stays as open as possible and that nobody is allowd to make a closed source (commercial) product with it. I belief it kills community effort if it's allowed to use parts from that open source to monetarize it in a closed source product.

DE-Koodie, 2019-01-16:
BTW i created a facebook group, as google+ is going to be deleted (sadly)
https://www.facebook.com/groups/938914776298161/

cprezzi, 2019-01-16:
@DE-Koodie I have deleted your post about the FB community, because we did not yet decide where to move and we don't that the community is split to multiple places!

DE-Koodie, 2019-01-17:
> @DE-Koodie I have deleted your post about the FB community, because we did not yet decide where to move and we don't want that the community is split to multiple places!

Oh sure, i wanted to help out and not to start confusing :) Sorry, did not really think about that.

kub1x, 2019-02-21:
@cprezzi What is the current primal place to ask questions? Google Plus is long dead (can't register to even ask a question), yet website of LW4 still points to the G+ page. Don't want to submit issues on Github when I just want to ask the community something.

cprezzi, 2019-02-27:
Our community is still active on G+ and will be moved to https://forum.makerforums.info

msurguy, 2019-03-14:
@cprezzi I just wanted to thank you very much for getting involved in taking over the Laser Web project and maintenance! 🥇 💯 👍

closed 2019-03-14 by msurguy