LaserWeb/LaserWeb4 #533: "The file has minor issues. Please check document is correctly loaded!", LaserWeb 4.0.996 error

asasergio, 2019-01-04:
Windows 10 (64bits)
LaserWeb 4.0.996
Inkscape 0.92.3

Files -> Documents -> Add Document...

[image]

after "Add Document"...

[image]

after Create Single -> Cut Rate -> Filter Stroke...

[image]

Have been tried with several other SVG files (Inkscape SVG and SVG flat) all resulting in the same error.

Not corresponding to [Youtube "LaserWeb4 Basics - Working with vectors"](https://youtu.be/_0jwzhZL4OU?list=PLCm8o-62Fu2nwrii2o1TTF5u1QC8zxgK0)

Something I did wrong?

cprezzi, 2019-01-07:
I guess your SVG has only fill color, but no stroke color. Try creating the operation without filter first. If that works, go back to Inkscape and check that stroke color is set.

asasergio, 2019-01-07:
Perfect,
I include the strokes in the figures generated by Inkscape and now I have the filter working, even though the LW4 continues to report the error.
Thank You very much.

[image]

The figure .svg created in Inkscape that can be downloaded for use in the "LW4 Basics - Working with vectors" tutorial...

[image]

cprezzi, 2019-01-08:
The message is just a warning, not an error. Most of the time it still works.

closed 2019-01-08 by cprezzi