LaserWeb/LaserWeb4 #535: Crash on lage SVG files

DE-Koodie, 2019-01-16:
Using Laserweb 4.0.996 the application crashes during import of very large SVG files.
The filesize is 9,14 MB (9.589.815 Bytes), a vector created with inkscape.

The workspace is basically empty, i just start laserweb and directly import the file.
CPU goes up to 16% and ram goes up to 1,7GB and then the app just whites out.
CPU usages drops and the window

[image]

DE-Koodie, 2019-01-16:
I don't want to publish the SVG im using as it has economical value, i can send it to some email confidentially...

DE-Koodie, 2019-01-16:
I checked if it has to do with a memory limit, but as i choose to raste the file it goes up to 790mb and crashes, too.
[image]
The bar is stuck at 100% for a while and then crashes the app again in the same white manner.
[image]

cprezzi, 2019-08-09:
For raster engraving, try to increase the tool with setting to something >= 0.2mm. This defines the "resolution" of the generated gcode lines (G1).