LaserWeb/LaserWeb4 #536: No supported firmware detected

Kgv1, 2019-01-25:
Hi

I have uploaded GRBL-LPC to a Smoothieboard and try to connect it to the latest version of LaserWeb4 on my computer running with Windows 8.1. When I connect to the board I get the message “Machine connected” but after about 10 second I receive the message “No supported firmware detected. Closing port COM3”. I have tried to save the .env file to the installation folder, also with “GRBL_WAIT_TIME=5” included in the text file (with several values for the wait-command). I included the LOG_LEVEL=3 command in the text file and it wrote a logfile in the installation folder. 

Unfortunately, nothing worked. I’ll really appreciate any help solving this issue.

Kgv1, 2019-01-26:
@cprezzi If it is not too much to ask, I would really like to know your thought on this issue. The firmware and software are the only thing left to finish my project. I have struggled with this for a long time now and cannot solve it on my own due to my limited knowledge and experience at programing and software.

cprezzi, 2019-01-28:
You already get my support for this issue on G+ (https://plus.google.com/u/0/117932834443228105559/posts/8GEFTiJ8kGf?fscid=z13kh3yawy2qy3hdf231v3oqmz3ajpklr.1548632507002535) so i'll close this issue here.

closed 2019-04-02 by cprezzi