LaserWeb/LaserWeb4 #537: Unsupported file type:application/octet-stream

CrispyCrafter, 2019-02-06:
- [X] Refers software
Binary build v4.0.996-134
[files.zip](https://github.com/LaserWeb/LaserWeb4/files/2836609/files.zip)

When reading in a SVG i get the following message:
``Unsupported file type:application/octet-stream``

The console reads as follows:
``The file has minor issues. Please check document is correctly loaded!``

I have included 3 variants of the same file,for comparison

cprezzi, 2019-08-09:
I don't get a "Unsupportet file" message when I load your files into LaserWeb. The "file has minor issues" is just a warning, nothing to worry about. 
I can generate the gcode without problems. But your SVG is 210x297mm (A4) which is bigger that the K40 work area and therefore the object is outside of the range. You should mark the object and only export the marked objects when saving the SVG (or set the page size to the work area of your machine).

CrispyCrafter, 2020-07-21:
thanks @cprezzi 
Didn't realise that would cause the issue.

closed 2020-07-21 by CrispyCrafter