LaserWeb/LaserWeb4 #538: Closed paths leaves tabs

a7hybnj2, 2019-03-03:
I cannot load an .svg into github but it looks like if you load the Workspace it includes that file.

My issue which you can see pretty clearly in the picture is that each of these cutouts has a tab holding it in. They are closed vector shapes and should be cut all the way through.

I have tried adding `M3 S1000 & M5` to my gcode tab to generate start and stops for the laser and turning off $32 but that just results in burned spots.

I have also tried varying the speed of cut and if I cut slowly with full laser power it burns the shapes out but also catches on fire. If I cut slowly but turn the laser intensity down it leaves tabs.

I have used software in the past that has a checkbox that allows the generated gcode to overlap the start of each cut slightly. (I forget what the checkbox was called but it is present in a flatcam fork I used). Is there anything similar to this that I can do in laserWeb to get clean cuts?

[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/2922687/Laserweb-Workspace.json.txt)
[image]

cprezzi, 2019-03-04:
What type of laser do you use (CO2, Diode...)?
Can you tell if the beam is starting too late, or switching off too early?

arthurwolf, 2019-03-04:
Laser cutting firmwares need to modulate power proportional to speed as
they accelerate/decelerate. It sounds like your firmware is either messing
that up, or has some setting/tuning problem around that feature.

a7hybnj2, 2019-03-04:
@arthurwolf I am running grbl and I am not aware of any tuning associated with 'laser mode $32'

@cprezzi it is a 2.5w diode. I will have to run some tests and compare paths to be able to tell if it is too early or late. I'll do that later today.

arthurwolf, 2019-03-04:
It could also be a problem with the diode's power output not being linear
relative to the pwm/ttl input signal meant to control power variation.

a7hybnj2, 2019-03-04:
@arthurwolf do you know of a way I can test this? Everything I have used the laser for prior to this seems to have worked as I expected it to. This is the first time that I have done something fine enough to notice issues.

When I use flatcam to generate gcode the closed vectors meet up perfectly but I understand how this could be firmware tuning of laser mode.

I am using a elekslaser 2.5w
$
Trinamic stepper ICs

cprezzi, 2019-03-05:
@a7hybnj2 On cheap laser engravers, sometimes the laser modules have integrated diode drivers and are only driven by switched 12V and GND signal (without separate PWM input). This is a bad, cost saving method that leads to  delays and non linearity. Can you please post a picture of your laser module and the electronics?

By the way: In grbl lasermode ($32=1), you have to use M4 instead of M3 to get the power relative to speed during accelleration/decelleration.

a7hybnj2, 2019-03-05:
@cprezzi I wasn't able to do any test burns but here are some pictures of the laser module.

[image]
[image]

cprezzi, 2019-03-05:
Ok, it has at least three pins, but the colors are not as I would expect! You should check that VCC is going to permanent 12V and PWM/TTL is connected to the Laser PWM Pin of the Arduino Board.

a7hybnj2, 2019-03-06:
Looks like it is hooked up correctly. I haven’t had time to do another test to see if it is starting late or stopping early.

[image]

a7hybnj2, 2019-03-07:
So, I am doing another burn today. Although I haven't been able to tell if it is starting late or stoping early I am getting tabs from settings I thought I wouldn't.

I thought originally I could just go full power and slow and have it burn all the way through. I am still getting tiny tabs. Like the width of 1/2 a hair.

Is there no way to tune the ramping that 'laser mode' does?

My next best idea is to write a script the re-processes the .nc file and copies the first few lines of each path onto the end of the path so that there is a little overlap. I don't really like this idea as 'the first few lines' isn't a specific measurement and could result in a lot of overlap or barely any.

cprezzi, 2019-03-07:
As long as you only need vector cut and no grayscale, you can try to add a delay command (G4) in the "Tool On" or "Tool off" settings.

For example, if the laser is starting too late, Try the following in "Tool On" (two lines):
M3
G4P0.5
(This would switch laser on and then wait 0.5s before executing the next command.)

a7hybnj2, 2019-03-09:
I haven't had the chance to try your latest suggestion but I wanted to make sure you wanted me to try it with `M3` and not `M4`...

cprezzi, 2019-03-09:
Yes, thats what I meant. LaserMode does more than just dynamic power adjustment, it also doesn't stop moving when the S value changes.

a7hybnj2, 2019-03-15:
Well, I was able to get my project done using M3 but their were still a bunch of 'danglers' I had to remove by hand. It was much better than using M4 though.

I will continue to test but now that this project is done I don't know when I will be lasering again.

closed 2019-04-02 by cprezzi