LaserWeb/LaserWeb4 #539: Windows 10 BSOD

tsipa, 2019-03-07:
Hello. I have Chinese board https://goo.gl/FgopMb similar to this one, running grbl 1.1f. I'm using windows 10 64bits to connect to this board using USB. After hour or two of continuous connection windows dies https://i.imgur.com/KqjO9qr.jpg complaining about CH341S64.SYS

I've read tons of report similar to this one, and every time chinese chip was considered guilty by default, but the thing is that if i run, let's say arduino serial monitor for hours or days it'll work fine wihtout BSOD, candle or lasergrbl also works as expected, so i suspect that if CH340 or driverl have problem at least there is workaround. I dig a bit around and found that node-serialport known to cause BSOD for another converter https://github.com/node-serialport/node-serialport/issues/1681

it seems like for most recent binary build available for windows we using 4.0.7 version of node-serialport, while this commit https://github.com/LaserWeb/lw.comm-server/commit/bdee9271dd83ccacb00d8b9a121d449ff1c3fdae bumped version 6.2.2

Could you please release new binaries with this commit, so i can check if BSOD problem is gone?

tsipa, 2019-03-08:
I can confirm that using latest version from master i cannot reproduce issue anymore. That would be much appreciated if you can release windows binaries - i've tried to build them on my own for a while without real success and now i'm about to give up.

pen25, 2019-03-24:
i decided to just install laserweb4 onto a pi3b if you have one laying around or can pick one up. id suggest doing it with the pi as you can leave the usb plugged in

tsipa, 2019-03-24:
I don't like the idea of solving software issue with bringing unnecessary hardware onto picture.

pen25, 2019-03-24:
Sometimes you just have to unless you want to out in the time to fix the driver issues. I'm sure you can post a bounty to get it corrected. Try it and you won't look back. Been using octoprint for several years and it's amazing people still use SD cards / tethered pcs

cprezzi, 2019-03-25:
@tsipa You can also solve the problem by using a decent controller board (with better USB driver), but I prefer the RasPi solution, because it's much more convenient to connect the machine over network.