LaserWeb/LaserWeb4 #540: No way to manually set serial port

padcom, 2019-03-30:
This is a simple one yet so frustrating...

I have a remote serial port exposed via ser2net. It doesn't show (obviously) in the list of serial ports, which is fine. What is completely awkward is that I can't type in the path to the serial port because the dropdown is fixed on the recognized serial ports only.

Please make it possible to type in the serial port name manually. All other applications allow for that - just this one doesn't.

cprezzi, 2019-04-01:
There is no official standard for "Serial over LAN". Usualy you need to install a driver in the OS to generate a virtual serialport for the network connected device.

LaserWeb can do "Serial over Telnet" for connecting network devices that are telnet compatible (like smoothieboard or Grbl-ESP32). To use this, you need to select telnet as the "machine connection", then you can enter the address in the machine ip field.

cprezzi, 2019-05-21:
Closed because of missing resonse,

closed 2019-05-21 by cprezzi

padcom, 2019-05-21:
I suggest you read the description of this thread again and rethink what you wrote. I know that on Windows you have to have a special device created (because well Windows is like that) but on other Unix-like operating systems it is just a file name.

ser2net being the de-facto standard serial port server and socat being the absolute standard to connect remote tcp/ip connection to a file in the filesystem does form the standard on unix-like systems. Not my fault windows is different.

Please allow to manually specify the name of the serial port - don't be smarter than the user. We know our computers better than you.

cprezzi, 2019-05-21:
If you know it better, then wy don't you fix it yourself?