LaserWeb/LaserWeb4 #542: OpenGL won't start. This app can't run without it. Windows fresh install

nincns, 2019-04-10:
fresh Installation of Laseerweb on Windows 10, OS system driver uptodate, NVidia GForce 1070ti try to install Laserweb on Nvidia Settings and let settings change by software, nothing changed

cant open Laserweb, allways showing 
[image]

nincns, 2019-04-10:
trying by start es administrator also not solve issue

cprezzi, 2019-08-09:
I guess you either don't have OpenGL drivers or they are not compatible with LaserWeb needs.
Please check if and what version of OpenGL your system supports. See https://community.esri.com/groups/technical-support/blog/2012/03/15/what-version-of-opengl-do-i-have-and-do-i-have-the-latest-drivers-for-my-system for details.