LaserWeb/LaserWeb4 #544: Laser Raster Issue on MAC OS (Never Ends)

txusballesteros, 2019-05-15:
### Follow this steps to open an issue ###
Hi everyone,

I'm newbie with LaserWeb and I'm plenty thankful with the work did by this community.

So I would like to report an issue that I'm having when I try to burn a raster Image. The problem is that I'm not able to complete the generate process because it never finish. I mean, the process comes to 100% but never finishes.

I'm using the version 4.0.993 on MAC OS
 

[image]

[Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/3181998/Workspace.json.txt)

cprezzi, 2019-05-21:
Did you scale the picture massively down in LW? If so, try to scale it down in a graphics program and load the smaller image in LW.

Try Laser Diameter 0.2mm. That only creates 1/4 of the data (compared to 0.1mm).

(By the way: the most recent version is 4.0.996)

txusballesteros, 2019-05-23:
Thanks a lot @cprezzi for your answer.

closed 2019-06-14 by cprezzi