LaserWeb/LaserWeb4 #545: Buffer overflow on Ardunio Nano with grbl 1.1f

harg505, 2019-05-21:
I am using lw.comm-server-4.0.130-x86_64.AppImage on ubuntu 18.04 with a ardunio nano v3 (genuine) running grbl 1.1f

with programs with lots of points such as this one:
[guitar.txt](https://github.com/LaserWeb/LaserWeb4/files/3200819/guitar.txt)

the Gcode sender in the machine control tab over runs the serial input buffer on the ardunio causing commands to be missed or misinterpreted.

looking at the grbl docs https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface

the same program sends fine using there simple_stream.py script from

https://github.com/grbl/grbl/tree/master/doc/script

cprezzi, 2019-05-21:
Are you sure it is a buffer overrun and not a serial communication problem?
Do you get an error message?

cprezzi, 2019-05-21:
I have run your gcode on my setup (LaserWeb EXE 4.0.996 on Win10, Server 4.0.115 on Raspberry Pi 3B+, Grbl-LPC on MKS SBase V1.3) and it worked without any problem. 

Then I tested with GRBL 1.1f on Adruino Nano (China clone with CH340) locally connected to Win10 with LaserWeb EXE (Frontend 4.0.996 / Backend 4.0.127) and also did not get any problem. (Both tests was made with 115200 baud)

It looks like your problem either has something to do with the Linux version or your setup.

harg505, 2019-05-21:
I have reproduced it on my works laptop which is a dell rather than ASUS
with another genuine arduinio both ar running  ubuntu 18.04 LTS,  but the
sender grbl simple sender script works on both.

harg505, 2019-05-21:
do you want my GRBL setup parameters?

cprezzi, 2019-05-22:
Yes, please post your $$ settings, just to be sure, but they should not be able to affect the serial communication. That all sounds more like a Linux specific problem.

Do you get any response in the console window of LW (bottom right) when the machine stops?

closed 2019-06-14 by cprezzi