LaserWeb/LaserWeb4 #546: RAMPS+Marlin Pins and Commands

FlailAway, 2019-05-27:
Hi, very new to all of this but I have an old 25W CO2 laser and the head movement electronics are toast but it still burns fine with 5KHz 5v PWM applied with a frequency generator. 

I have a Mega+RAMPS, currently Marlin 1.1.9 isntalled. The Marlin instructions say to enable the Laser and map PWM to pin D6 and I am getting a good signal there. Marlin also says to use "M107 P1 S0" for OFF and "M106 P1 Sxxx" for intensity. Both work fine for OFF and varying intensity.

I found some 2018-help here but the instructions for OFF and ON are M116 P0 etc. I am unsure what to do with that since Marlin is working fine with M106... for the PWM.

I am OK with going to grbl-Mega if that would be better, but that wiki and cpu_map.h files do not agree on PWM frequency and/or pin assignments. I really don't care what firmware I use as long as I can get the old laser moving again.

Could someone please point me at some instructions on how to get this Rastering and cutting again?

Thank you.

cprezzi, 2019-05-28:
Marlin was never optimized for laser operation and has changed the way it works several times. LaserWeb needs some configuration to generate working gcode for Marlin. 

Go to Settings / Gcode and set the `GCODE GENERATOR` to `marlin`. Then set `TOOL ON` to `M106 P1`, `TOOL OFF` to `M107 P1 S0` and activate `INTENSITYY SEPARATE LINE`. This should generate gcode that works for marlin.

cprezzi, 2019-05-28:
Hint: Working with Marlin is a compromise. The results, especially for grayscale engraving, will not be as good as with oter firmware (like Grbl, TinyG or Smoothieare). This is caused by missing PWM adaption on accelleration, decelleration.

FlailAway, 2019-05-28:
> Working with Marlin is a compromise.

Thanks and I'd really prefer to be using your adaptation of grbl, b the LCD+SD card is needed as there is no computer down in the garage.

Is there something else I can use for LCD+SD card? I have an MKS-SBASE board that I installed your  1.23-port-version on but the TTL was only 3.8v-Peak-to-peak. So was not sure that would be enough for the 5v-P-P the old (1998) Synrad laser expects. 

Using the MKS, could I use the TFT32 LCD with the 3D printer menus?

cprezzi, 2019-05-28:
If 3.8V is not enough for your laser PWM signal, then you just need a cheap logic level shifter (like https://www.adafruit.com/product/757), or you make your own transistor/mosfet voltage switch. 

As far as I know Smoothieware cannot handle the MKS TFT32 LCD, only the usual RepRap full grafic LCDs. grbl-LPC doesn't support any displays.

FlailAway, 2019-05-29:
Thanks for the follow up. 

The MKS TFT works fine with the SBASE board and Smoothie firmware. The TFT firmware has a "Smoothie" option and you also set the smoothieware for the graphic controller. But, the smoothieware software was so terrible to use, I have gone back to the MKS-Gen-L and TFT with Marlin on the Delta printer.

With the MKS-SBASE freed up, I thought the extra speed might be nice for the laser. I have now tested it on the old laser and while the 3.8v works, it does not do so reliably. I am assuming that is because of the variation in that 3.8v sometimes dipping close to or below 3.5v.

As to the level shift, I have some [like these](https://www.amazon.com/dp/B07F7XCHHN)

I now have grbl-Mega working reliably at 5KHz and 5v-P-P with the RAMPS board. Should I stay with that or keep going with making the SBASE+level shift work? 

Meaning, is the Arduino Mega up to the task of running the laser fast enough?

Thanks for your time so far.

cprezzi, 2019-05-29:
I never tried grbl-Mega but in theory the MCU (8 bit) is much slower than the 32 bit MCU of the SBASE board. I would go with the SBASE and grbl-LPC, because this will deliver much faster raster engraving speed.

cprezzi, 2019-05-29:
The level shifter from your link should work just fine.

cprezzi, 2019-05-29:
Here my maximum raster feed examples (same picture & settings):

- Adruino Uno (ATmega328p) with grbl 1.1f up to 35-50mm/s 
- MKS SBASE (NXP LPC1768) with grbl-LPC more than 300mm/s

FlailAway, 2019-05-29:
>     * MKS SBASE (NXP LPC1768) with grbl-LPC more than 300mm/s

OK, thanks, no question that I should be using the SBASE. Dang, just spent the best part of two days getting the RAMPS working. All good experience I guess .

FlailAway, 2019-05-30:
> The level shifter from your link should work just fine.

They are not fast enough at 5KHz when going 3v to 5v. On the 'scope, At anything over about PWM-5% I get a solid 5v line with a lot of noise. I will order one of the Adafruit ones, but suspect they may be no better.

cprezzi, 2019-05-31:
Hm, that sounds weird. They are made for I2C, which goes from 100kbps up to 3.2Mbps!

closed 2019-06-14 by cprezzi