LaserWeb/LaserWeb4 #547: node server.js working, service runs, but frontend can't connect to the server if server.js isn't started manually

chismah, 2019-05-27:
Hi.

I get a Pi 3B+ to run a server for laserweb, and so I am not really skill when it comes to code, but I try my best with info gathered online..

I followed the wiki to install the server on raspian.

I properly get the server.js running with  the `node server.js` command.
But it seems odd I saw the message : _use http:// 127.0.1.1:800 to connect this server_ while I set a static local ip to 192.168.0.50 editing etc/dhcpcd.conf (`interface wifi0
static ip-address=192.168.0.50/24
static routers=192.168.0.1
static domain_name_servers=192.168.0.1`

the `systemctl status lw.comm-server` command indicate that the service is running

In the end, when I boot the Pi, the lw.comm-server service is running, but I can't connect to the server with the frontend (Mac OS).
But if I launch manually the server with the `node server.js` command,  I am immediately able to connect the pi at 192.168.0.50:8000 (even though the server.js tell me to connect to the pi at 127.0.1.1:8000).

I tried recreate the service files, but it ends the same.

For now I launch `node server.js` through ssh.
Connection is lost as soon as I end the script with Ctrl+C

Does anyone add the same issue?

Regards

cprezzi, 2019-05-28:
lw.comm-server is serving a website and websockets on the same port (8000).

If systemctl tells you the service is running, try opening url 192.168.0.50:8000 (the IP of your raspi) with a chrome browser on your mac (the frontend needs chrome to work properly). Hint: This frontend is older than the version in the installers for OX, Win.
If this works, you should also be able to connect with the installed version of LW4. You only have to change the server adress (in connect tab) to the same ip:port.

cprezzi, 2019-05-28:
Also check if the service has a permission problem or a wrong path in the script.

chismah, 2019-06-03:
I chmod777 the service, paths don't seem to point nowhere. (there is a node file in /bin and the server.js in /home/pi/lw.comm-server/)
```
[Unit]
Description=LaserWeb4 server

[Service]
ExecStart=/usr/bin/node /home/pi/lw.comm-server/server.js
Restart=always
RestartSec=10                       # Restart service after 10 seconds if node service crashes
StandardOutput=syslog               # Output to syslog
StandardError=syslog                # Output to syslog
SyslogIdentifier=lw.comm-server
WorkingDirectory=/home/pi/lw.comm-server
User=pi
#Group=<alternate group>
Environment=NODE_ENV=production PORT=8000

[Install]
WantedBy=multi-user.target
```

Issue remains the same.
I can't connect through the frontend (nor with chrome or safari) without running `node server.js` first.
> Node version is v10.15.3
> Nom version is v6.9.0

Regards

JHarding86, 2019-06-04:
I am having this same issue on my pi 3b+. systemctl says the service is running right after a fresh boot, but when attempting to connect to the front end via chrome, there is no response, or when I use the Windows GUI and attempt to connect to the server... nothing. 

I must manually start the server with
`node server.js`

I have previously checked that the script is pointing to all of the correct file locations. When I get home tonight I will look at the permissions and see if there is an issue.

cprezzi, 2019-06-05:
Please use https://forum.makerforums.info/c/laserweb-cncweb for further assistance.

closed 2019-06-14 by cprezzi