LaserWeb/LaserWeb4 #548: [LaserWeb 4 - bug] needs restart after grbl update

FlailAway, 2019-05-29:
[bug]
Hi, not a major issue, but I am getting grbl-Mega to work on a RAMPS board and had the need to upload (using Arduino-IDE) slight changes to the cpu_map.h etc files, many times. 

After an upload I would reboot the Mega but LW would connect for about 10-seconds then return "no firmware found" and disconnect. I tried everything, (disconnect, reassign the USB etc) I could think of but the only way to test-cut was to shut it down and restart. That meant reloading the test-files etc.

It would be nice if LW did a complete restart with the connection after a "Disconnect" and a subsequent "Connect." Once the entire LW-program was restarted it was happy with everything again.

cprezzi, 2019-05-29:
LW is checking the firmware version by reading the welcome string that is comming from grbl after each connect. If the welcome string "Grbl 1.1f ['$' for help]" (1.1f or higher) is not comming or different, then the connection is shut down because the firmware doesn't seem to be compatible with LW. 

Is gebl-Mega sending a different welcome message?

There is also an option to send a reset command after connect to force grbl to send the welcome string. To activate that, you need to create a textfile called ".env" (nothing in front of the dot!) with the content "RESET_ON_CONNECT=1".

FlailAway, 2019-05-29:
> Is gebl-Mega sending a different welcome message?

Yes, "grbl 1.1g" and I had wondered about that, but, it works OK if I shut it down and start again and then it is happy with that 1.1g message.

cprezzi, 2019-05-31:
That's why I wrote the option of RESET_ON_CONNECT. This makes sure that the Arduino is reset on each connect.

cprezzi, 2019-05-31:
But you should manually reset your arduino anyways after flashing the firmware.

closed 2019-06-14 by cprezzi