LaserWeb/LaserWeb4 #550: Mill gcode doesn't create

Gtumer1992, 2019-06-13:
Good day to everyone! I got a trouble with laserweb mill gcode generation. I made a model in sketchup and export it in dxf format. After the model appeared in laserweb, I set settings in the fields (like depth pass, tool diamiter and so long) and push generate.
The programm write "Ellapsed: 00:00.345", but no gcode generates. Sometimes it generates gcode with same model, but I can't catch why it happends in such way. 
[image]
[image]

cprezzi, 2019-06-14:
Please go to https://forum.makerforums.info/c/laserweb-cncweb for user support.

closed 2019-06-14 by cprezzi

alubaba95, 2019-06-14:
Rryhjj