LaserWeb/LaserWeb4 #552: Origin Top-Left Corner with Y moving in positive

adammhaile, 2019-08-11:
I have a custom built laser engraver and am trying to use LaserWeb with it. I set it up with the origin being in the Top-Left (some might say Back-Left) corner of the machine and with all coordinate values being positive from there. I also tweaked the Grbl firmware to enable setting the machine origin location to actually being 0,0 instead of some offset, after homing. This just played nicer with other software I'm used to using. However, the options in LaserWeb seem lacking for setting the origin location.

I tried setting "Machine Bottom Y" in the machine settings, as suggested in the docs. However, that puts the entire Y axis space in the negative and my Y axis is actually moving in the opposite direction to that. So any Y movement through the interface results in going out of bounds (I have soft endstops turned on).

[image]

Is there some way to tell it that the Y axis is inverted from what it seems to think is default? I've used lasers that home to both the front and the back left. So it's not like there's a real standard. But not very helpful if I can't get LaserWeb to understand how my particular laser is setup :(

tbfleming, 2019-08-11:
LaserWeb doesn't support an upside-down Y axis.

closed 2019-08-11 by tbfleming

adammhaile, 2019-08-11:
@tbfleming - That is unfortunate. No reason that's not a valid coordinate system on some machines. Please consider adding it.

cprezzi, 2019-08-14:
@adammhaile Having right, back and up as positive direction is a worldwide standard. There is no reason to change that. Just swap the direction of your y axis and all is ok. 

I also have my homing at top left, but made GRBL set the origin to bottom left. That works just fine.
Be aware that LW4 shows work coordinates not machine coordinates and you can set 0,0 where ever you like!

adammhaile, 2019-08-14:
@cprezzi - it would be nice to have the docs a little more clear on what that entails. New to GRBL so it took a while to figure how to do that (which I had done already). Also... seems like it's more accurate to say that LW4 expects GRBL to be reporting Work coordinates... because of some other software I was using it was set to machine coordinates which was screwing with things.
For anyone reading this in the future, what I did was:

- Set `$23=1` (home top/left - or back/left if you prefer)
- Made sure my motors were connected such that homing actually moved towards the "top", what I consider the "back"
- Run `G10 L2 P1 X0 Y-130` - I have a 130x130mm working are, so this is saying that 0,0 in the work coordinates is where it homed too minus 130mm. Making the homing position 0,130
- In LW4, set "machine bottom" to -130 as well.

cprezzi, 2019-08-14:
And, as you wrote, set $10=0 to get work coordinates from GRBL!