LaserWeb/LaserWeb4 #554: Lathe style rotary raster operation

infamous-panda, 2019-08-22:
Hello I was referred by Cprezzi to post this here. ATTN: tbfleming

I am making cylinders with a raster engraving that wraps around the surface seamlessly 360°. It would be really helpful to have have a mode where the motion acts more like a lathe. The A axis would move continuously in one direction (not back and forth) while the X axis slowly moves away. I have a setup that requires that the laser head be fixed and the workpiece moves. Currently I have to accelerate and move very slowly due to the mass involved. 

I'd be willing to contribute monetarily to develop this if anyone is interested.

cprezzi, 2019-08-23:
@tbfleming This is probably for you. What du you think about it?

tbfleming, 2019-08-23:
It's an unusual laser machine which has that limitation. I don't intend to develop for that case.

infamous-panda, 2019-08-23:
Hello Todd,

Thank you for responding. It is unusual, but I think it would also benefit
traditional setups whereby reducing the the back and forth may increase the
speed and reduce wear and tear.

 Regardless, Can you point us to anyone qualified and who may be interested
in developing this feature on a freelance basis? We would love to talk to
them.

tbfleming, 2019-08-23:
Only @jorgerobles and @cprezzi

infamous-panda, 2019-12-05:
How about just a 90 degree or variable raster option in addition the "diagonal" 45 degree ?