LaserWeb/LaserWeb4 #556: How to deal with resolution issues when generating bitmaps (to avoid eternal loading signs and freezing)

jdargot, 2019-09-01:
Issue so far:

Open LaserWeb 4.0.996
Select a 640 x 420 .jpg resized outside or inside the LaserWeb program.
(Alternatively open a 1024 x 684 .jpg without any alterations)
Select image
Choose "Create single" and set as follows:
- Laser Diameter: .2mm
- Min Laser Power: 5.75
- Max Laser Power: 12
- Cut Rate: 2000
Select Generate

Result:

Instead of completing the gcode, the program reaches 100% and then freezes. Nothing can be done until the program is restarted.

Comments:

I have successfully engraved a few pictures using lower resolution small images that I import and DO NOT ALTER from their original state. However, if I make any changes to resolution or image size before or after importing, this issue comes up. It is particularly problematic, because I am looking to engrave an image that is 300mm wide and the program just doesn't seem able to handle it.
I tried lowering the DPI in file settings, but this only seemed to make things worse. 

Questions:

Is there something in the settings that overcomplicates the process of generating a larger bitmap? If not, what are the requirements for an image to work on this program? 

Thank you,

cprezzi, 2019-09-02:
What is your CPU, RAM and OS?
Can you post one picture to try on my installation?

cprezzi, 2019-09-02:
When I use a 4000x3000 pixel picture and 0.1mm laser diameter (which equals to 400x300mm), it renders to 100% in about 90s and then I get a white screen hangup. 

When I scale the picture down to 300x200mm in LW and generate the gcode with 0.2mm laser diameter it creates the gcode correctly in under 30s (it works). The LW Process uses 1.6GB of RAM for this!

So I guess the problem is memory related.

jdargot, 2019-09-02:
Hmm, interesting. I'm running Windows 10 64 bit. My CPU is an i7 and I have 8 gigs of ram. I would have assumed that would be enough for anything. Even checking the details while running, it doesn't seem like the computer is working that hard. (That's another thing, I've played around with the generation threads and that hasn't changed it either). 
Here is one picture I tried. I share this one because I couldn't get it to work with or without resizing/changing resolution. 

[image]

jdargot, 2019-09-02:
Something else I will add: messing with the resolution, I can get that picture to work at 1/8 the resolution and a length of 100mm. The problem is that at 1/8, the resolution is pretty depressing for that size. And again, I am aiming for 300mm which it still will not process. Even if it did process it correctly, the picture would be so low resolution, you could hardly tell what it is.
Are there any other tricks for smoothing the process and still having a tangible quality?
Thanks again,

cprezzi, 2019-09-03:
That picture is only 1024x684 pixel. Is that correct or was it scaled by the forum upload?

cprezzi, 2019-09-03:
My PC is an i7 with 24GB RAM and Win10/64.

arthurwolf, 2019-09-03:
Weird flex but ok.

jdargot, 2019-09-03:
Yes, that is correct. 1024x684 is what's giving me the errors. It doesn't seem like it should be having those issues given the resolution combined with current processing power. Just to double check, I will download laserweb onto my other computer tonight and see if it does any better. Although my secondary is an i3 with 4 gigs ram, so I'm not expecting much out of it.

jdargot, 2019-09-07:
Interesting. So I tried it with my other computer (the one with half the processing power) and it generated with regular resolution in about 60 seconds. I don't know what could cause such a difference like that.

jdargot, 2019-09-10:
I figured out what it is (kind of). It worked on my new computer because I had forgotten to set the gcode to Marlin format. I noticed in comparing the codes that Marlin uses nearly twice the length because the laser on/off commands are on separate lines from the x/y coordinates. Sound plausible so far? The marlin settings only allow for small images and very low resolution. Is that common or unique to my computer's processing speed?
On a positive note, I did manage to modify the default gcode to work with marlin. I would really love to get these images working on their own though. 
Thanks,