LaserWeb/LaserWeb4 #559: Is ESP8266 connection possible?

jwidht, 2019-09-05:
Hi, 

In the LW4 connection options (MACHINE CONNECTION), it shows ESP8266 as one of the options.
I would like to utilize ESP8266-01 to connect to my laser engraver via WiFi (because, otherwise, there are too many wires around and I am getting to inhale too much smoke ;) )

 I tried following the instructions for LW3 here
[https://github.com/LaserWeb/deprecated-LaserWeb3/wiki/Connecting-ESP8266---LaserWeb3---Smoothieware](https://github.com/LaserWeb/deprecated-LaserWeb3/wiki/Connecting-ESP8266---LaserWeb3---Smoothieware)

also see [here](https://github.com/LaserWeb/LaserWeb3-ESP8266/issues/6)

In my case, I can see that ESP8266 with the ino sketch inside boots and connects to my WiFi network, I am also able to ping its IP address. However, when I open LW4 and select

MACHINE CONNECTION: ESP8266
MACHINE IP: 192.168.1.74
 (which is the ESP8266 address)

and press CONNECT, I am getting the following output:

> Connecting Machine @ ESP8266,192.168.1.74
>  Server error: unexpected server response (404)
>  Machine disconnected
>  Machine disconnected

Am I doing it right, is it possible at all? Has anybody managed to connect remotely, what did you use?

Thanks so much

PS: My machine is controlled by Arduino UNO with a stepper driver shield with the latest GRBL

cprezzi, 2019-09-05:
This connection was just a test and needed a special firmware on the ESP8266. Unfortunately it never worked reliable, because WLAN connection was not stable and fast enough.

One option would be to use a raspberry pi with the lw.comm-server installed on it. lw.comm-server is is a gateway between websocket (LAN) and machine (USB).

Another option would be to replace your Arduino/Grbl/CNC-Shield with a Grbl-ESP32 Board from Bart Dring (see https://github.com/bdring/Grbl_Esp32 and https://www.tindie.com/products/33366583/grbl_esp32-cnc-development-board-v35/).

jwidht, 2019-09-05:
Thank you, you are awesome!  👍 
I already have a Raspberry Pi Zero W which streams video to LW4, and it works great. I'll be looking into adding another one for the connectivity. 
Warm Regards