LaserWeb/LaserWeb4 #560: SVG layers not pulled in correctly

AllenSlaghuis, 2019-09-18:
i have picked up that not all my layers are pulled in to Laserweb 4.
it gives me errors in the console at the right. i am using it on a macbook. and am using photoshop cc to export to *.svg format. not sure if i am doing something wrong.

The Error : the file has minor issues . please check document loaded correctly

i had to change the svg to .txt to upload it.

[image]
[image]
[FCE.json.txt](https://github.com/LaserWeb/LaserWeb4/files/3624826/FCE.json.txt)
[test.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/3624829/test.svg.txt)

cprezzi, 2019-09-19:
Photoshop is not realy known to generate good SVGs, as it's not a vecror program. A possible solution is to save the file as a JPG. This way the left vector part is converted into a pixel image and all can be engraved (laser raster) together.

AllenSlaghuis, 2019-09-19:
Thank you for the feedback. then I have 3 questions.might be the wrong place to ask them.
1) what would you segest then I use , Inkscape ?
2) the layers are for raster , detail cut & then finel outside cut ?
3) I trede Inkscape but it also gives me the same problems ?

Thank you

cprezzi, 2019-09-19:
Inkscape is a good free solution. You just need to be aware, that colored infill doesn't meen anything in vector data, as the laser can only follow lines. If you want to fill an area, you need to fill it with a lot of lines. This can be done directly in LaserWeb with the [laser fill] operation, or via a hachfill plugin for Inkscape.
Alternatively you can convert such a picture into a pixel grafix and use the [laser raster] operation, which scanns horizontaly over the graphics and only burns dark points.

Instead of layers, I would recommend using color filters to separate operations in LW.

cprezzi, 2019-09-19:
If you want to engrave and cut in one job, you can import a JPG (for the engrave) into Inkscape and add the vector shape for the cut. Then save as SVG. LaserWeb can load both pixel and vector data from the same SVG.

AllenSlaghuis, 2019-09-20:
Thank you for your help. this helped me allot. will have to teach my self Inkscape.

AllenSlaghuis, 2019-10-03:
So i have been batteling and still end up with the same problem. i have installed inkscape . traced the things i want to engrave but. still only parts of it get puled in to LW. not sure what is going on ?
here is the screen grab from laser web :

[image]

cprezzi, 2019-10-04:
I don't understand why you need to vectorise (trace) the image in inkscape. You could directly load a JPG into LaserWeb and engrave it using the "laser engrave" operation.

SunnyadiDigital, 2026-08-20:
If SVG layers are not importing correctly, check whether the artwork was properly structured before export. Ungrouping elements, preserving paths, using clear layer names, and exporting with compatible SVG settings can help. For complex artwork, professional SVG conversion can ensure clean, editable, production-ready files. **Vector Tracing Pro** provides manual [SVG conversion services](https://www.vectortracingpro.com/pages/svg-conversion-services) for logos and artwork, with files prepared for different design and production needs: