LaserWeb/LaserWeb4 #561: Cannot generate G-code in negative dimensions.

yoch2015, 2019-09-24:
Hello.
If ESP32_grbl is used but the dimension is set to a negative value because the work position is outside the machine setting range, the g code generation button is prohibited and cannot be generated.
What should I do now?
[image]

cprezzi, 2019-09-25:
There is nothing like a machine with negative size! These values must be positive.
They have nothing to do with the machine origin or work origin.

What exactly do you try to achieve?

yoch2015, 2019-09-26:
I set it on the wiki.
[https://github.com/LaserWeb/LaserWeb4/wiki/3.1-Settings-Machine#machine-width--height-values-will-determine-the-machine-workable-area-this-will-be-represented-by-4-arrows-and-a-highlighted-white-area](url)
If you turn on SHOW MACHINE with a positive value, the frame will come out of the machining area.
I just want to match it.
[image]

cprezzi, 2019-09-26:
This problem is caused by the homing logic of grbl. By default, grbl's homing sets the machine in negative space, which is usual for CNC Mills.

Did you configure $10=0 (so grbl reports work coordinates)? Have you tried to move to bottom left and "set zero"?

I have made a grbl fork with an additional config switch to home to positive space (and also a switch to use PWM for a RC-servo). Please check https://github.com/cprezzi/grbl-servo for details. You would need to disable the servo switch by commenting out line 377 (#define SPINDLE_IS_SERVO) by adding two // in front.

cprezzi, 2019-09-26:
I have made a fork of ESP32_grbl with my HOMING_FORCE_POSITIVE_SPACE feature. You can find it @ https://github.com/cprezzi/Grbl_Esp32

yoch2015, 2019-09-27:
$ 10 = 0 is set.
Build with forked source.
I am satisfied with the results I wanted
Thank you very much.
[image]

closed 2019-09-27 by yoch2015

cprezzi, 2019-09-30:
You are welcome. :)