LaserWeb/LaserWeb4 #569: Shift problem with an EleksMaker

ricochebonjour, 2019-10-09:
Hello everyone, do some of you use an EleksMaker Pro (62x62cm) with LaserWeb 4?

I have shifts when I want to make a laser fill) + laser cut, I tried different settings (laser power, speed, etc.) in the settings but it persists. Sometimes the alignment of the letters of the word is missed too.

A picture is worth 1000 words :
[image]

Do you have any idea where it can come from ?
Otherwise, what settings do you use in LaserWeb with your EleksLaser ?

Thank's so much

cprezzi, 2019-10-09:
What are your feed and acceleration settings? 
I gess you loose stepps somewhere. Probably caused by too high accelleration.

ricochebonjour, 2019-10-09:
My settings, based on a post from EleskMaker forum : 

$100=80.000
$101=80.000
$102=80.000
$110=8000.000
$111=8000.000
$112=0.000
$120=5000.000
$121=5000.000
$122=0.000

What should I change ?

cprezzi, 2019-10-09:
Your acceleration of 5000mm/s is a bit fast. I would suggest to begin with 1000mm/s. You should also check your stepper current. 
(check https://laserweb.yurl.ch/documentation/initial-configuration/31-grbl-1-1e for setup instructions)

ricochebonjour, 2019-10-09:
Ooops indeed, the good value is 500 and not 5000...
Anyway, I will follow your guide and check stepper current to adjust these, hope it will fix my problem !
I have to wait until sunday to check at home.

Thank you,

ricochebonjour, 2019-10-15:
Problem solved !
Perfect with $120=500 and $121=500 instead of 5000.

Merci Claudio,

closed 2019-10-15 by ricochebonjour