LaserWeb/LaserWeb4 #576: linkinfy "easy installer" at https://laserweb.yurl.ch/

matkoniecz, 2019-10-31:
On reading https://laserweb.yurl.ch/ I encountered "easy installer" words. It would be great to make it clickable link leading to this installers to make it easier to find them.

I will find them, but making it easier for new users would be likely a good idea.

Note that https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/.github/CONTRIBUTING.md is not mentioning where is repo for website.

matkoniecz, 2019-10-31:
Especially as from quick check I was unable to find Linux installer (or preferably - ppa with a package, so I can autoupdate it)

cprezzi, 2019-10-31:
@matkoniecz I hope you found the big `Downloads` button in the menu ;)

I have added a direct link from "easy installer" to the github binaries too.
There is no repo for the website, as it is a CMS managed solely by me.

The last linux "installer" was an AppImage here: https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/untagged-4818330b6baa8213d4a7

closed 2019-11-20 by cprezzi