LaserWeb/LaserWeb4 #579: Dont take off the laser on the blank space, please help

Milu1980, 2019-11-19:
### Follow this steps to open an issue ###

- [ ] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.

cprezzi, 2019-11-20:
Closed because of missing details!

closed 2019-11-20 by cprezzi

Milu1980, 2019-11-20:
iam working, i wanted at the night posting the pictures