LaserWeb/LaserWeb4 #581: LaserWeb.Setup.4.0.996-136

nino165, 2019-11-27:
Hello
I have been using this program for a long time and since I had the LaserWeb 4.0.996 version it worked.
Today it has installed the new version and it doesn't work anymore, the white picture remains and does nothing.
The installed version is LaserWeb.Setup.4.0.996-136

nino165, 2019-11-27:
[image]

nino165, 2019-11-29:
Ho installato la versione installata è LaserWeb.Setup.4.0.996-134
e funziona, bisogna rivedere perchè la 136 non funziona

adigandhi, 2019-11-30:
One more report for the same bug

davidmoreland, 2019-12-01:
I was using 134 because 136 won't hang at white screen.  However, now 134 does the same thing :(.  Using Windows 10 Home.

mdima77, 2019-12-03:
The same

Evg33, 2019-12-03:
The same

cprezzi, 2019-12-03:
Did you accept the firewall rules to allow LaserWeb? If it does ask for permission anymore, go to the firewall and add LaserWeb4 as trusted.

cprezzi, 2019-12-03:
I am able to replicate this problem, so I will try to find a solution. Please be patient.

nino165, 2019-12-04:
Hi
I went to the firewall and added laserweb as trusted, but the screen is always white.

cprezzi, 2019-12-04:
@nino165 Yes, it seems to be a different problem.

cprezzi, 2020-08-14:
For those who get a white screen please try deleting als LaserWeb settings:

```
Windows:
C:\Users\<user>\AppData\Roaming\lw.comm-server

Linux:
/home/<user>/.config/lw.comm-server

OS X:
/Users/<user>/Library/Application Support/lw.comm-server

!! REPLACE <user> WITH YOUR USERNAME !!
```

To clean all data you just have to delete the lw.comm-server folder.

jiahuahu, 2021-01-06:
这个问题如何解决我想知道答案