LaserWeb/LaserWeb4 #582: Laser not firing at Full Power?

UrsidaeSVG, 2019-12-05:
Hey there,
now since a few months i got a 3018Pro CNC with a 7W 450 nm Blue Laser. Many tests later it seems that i can't cut with full Power in LaserWeb.

If i use "EngraverMaster" the 4mm Beech Plywood cuts perfectly through in less cycles.

Settings i use:
100% Power
6 Passes
0,6mm Pass Depth
2mm/sec Cut Rate

With these settings i created a .nc file and tested it on EngraverMaster with these Results:
[image]

Here are my GRBL Settings:
` $$
 $0=10
 $1=254
 $2=0
 $3=2
 $4=0
 $5=0
 $6=0
 $10=0
 $11=0.010
 $12=0.002
 $13=0
 $20=0
 $21=0
 $22=0
 $23=0
 $24=25.000
 $25=500.000
 $26=250
 $27=1.000
 $30=1000
 $31=0
 $32=1
 $100=800.000
 $101=800.000
 $102=800.000
 $110=2500.000
 $111=2500.000
 $112=1500.000
 $120=160.000
 $121=160.000
 $122=48.000
 $130=300.000
 $131=180.000
 $132=44.000`

In LaserWeb i set the Max Power to 1000 and tested with 255 and 9999 with no change.

Any ideas what could be the issue?

arthurwolf, 2019-12-05:
Are you sure the Z doesn't go to a different position in each case/the
focus is the same ?

UrsidaeSVG, 2019-12-05:
Jep, focus is always checked before lasering.

arthurwolf, 2019-12-05:
Are you sure they really go at the same speed, etc ?
Maybe post the first 100 lines of each file to pastebin.com and give us the
links here ?

UrsidaeSVG, 2019-12-05:
I use the same file for both programs. I set up the task in LaserWeb and saved it as .nc.

https://pastebin.com/y8gsS8hD

Double Checked the Speed. It's 120mm/min in EngraverMaster and 2mm/sec in LaserWeb.

cprezzi, 2020-02-11:
What controller and firmware are you using?

UrsidaeSVG, 2020-02-14:
Arduino Uno and GRBL 1.1

cprezzi, 2020-02-17:
Can you please try if it make a difference when using M3 instead of M4?
I guess the adaptive laser PWM limits the power you reach, because the line segments are short and your accelleration is low.

UrsidaeSVG, 2020-02-25:
Tested and it did not really make a difference.

Madcreatorzz, 2020-04-05:
I was having problem like yours anf found my solution today. (Maybe it's not that but worth telling if it resolve your problem...). 
My problem was the opposite. When at 2%, the laser was almost at full power. I found that my GRBL pwm settings was completly wrong (my PWM frequency was at 2 Mhz when my laser was expecting 20Khz). So I changed the GRBL board settings to generate a proper PWM frequency for the laser. Maybe you could try to look for this...