LaserWeb/LaserWeb4 #583: Laser raster stripped lines

kcl93, 2019-12-06:
Hello :)
I am currently using LaserWeb to cut and engrave some thin wood using my retrofitted 3D printer.
While Laser Cut and similar work fine I have an issure generating correct G-Code when trying to raster an image. To control my output power I use a command such as "M106 P1 Sxxx". The problem is that this line will replace it with a "stripped line" comment:
https://github.com/LaserWeb/LaserWeb4/blob/218e87e7c69933bcf7d600631b68d71714984a2e/src/lib/cam-gcode-raster.js#L84
I guess that there is a reason for this functionality but would it be possible to check for other, more precise things other than just whether the new line starts with an "S"?
I could move my the M106 P1" part of my command to the Tool ON section but this leads to additional M106 P1 commands before each Path when using Laser Cut (M106 P1 is considered by Marlin the same as M106 P1 S255).
Just changing the condition for the above if statement might be the clean way to solve this problem (I would propose some code changes but I don't now the current reason behind this statement).
Thanks in advance!

Regards,

Kai

cprezzi, 2019-12-06:
There is no reason for code changes. It works as it should. Just make a machine profile whith theses settings:
[image]

cprezzi, 2019-12-06:
If you don't like the additional M106 P1 in the cutting code, you can make a separate machine profile for cutting with the following settings:

[image]

kcl93, 2019-12-06:
Hello cprezzi,

Thank you for your quick answer! This would indeed work if I only engrave or cut at a time. Regularly however I would like to do both at the same time (engrave an image then some text and then cut all of it). This would not be possible without having to start generating and start multiple files.
Changing the if statement however would make it possible to use the same profile independent from which kind of engraving I want to do.

Kind regards,

Kai

cprezzi, 2019-12-09:
I'm not the developer of this part, but I think this was implemented because a line starting with S is not valid Gcode.

closed 2020-02-11 by cprezzi