LaserWeb/LaserWeb4 #584: No supported firmware detected. - GRBL1.1f

Stady234, 2019-12-14:
Not sure what is going on...  I am trying to connect to my new laser... the default LaserGBRL program connects and I can burn images... when i try to connect via LaserWeb, it says connected for about 3-5 seconds then Closes port and states "No supported firmware detected."

[laserweb-settings.json.txt](https://github.com/LaserWeb/LaserWeb4/files/3963426/laserweb-settings.json.txt)
[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/3963427/Laserweb-Workspace.json.txt)
[laserweb-snapshot.json.txt](https://github.com/LaserWeb/LaserWeb4/files/3963428/laserweb-snapshot.json.txt)
[laserweb-profiles.json.txt](https://github.com/LaserWeb/LaserWeb4/files/3963429/laserweb-profiles.json.txt)

cprezzi, 2019-12-14:
The reason for that is, that LaserWeb checks for firmware compatibility (which other software don't) and if it doesn't receive a known welcome string a few seconds after connecting, it will close the connection again. 

Which controller board and firmware are you using?

Stady234, 2019-12-14:
Looks like it is a SMT32(or STM32) board with GRBL1.1f on it...

cprezzi, 2019-12-16:
I didn't know GRBL was ported to STM32. The question is, which welcome string is sent?

Can you please connect the board to a serial terminal (like the one integrated in Arduino IDE) and tell me the string you get after opening the serial connection?

Stady234, 2019-12-17:
Using ArduinoIDE I only got this ::
```
20:58:23.304 -> ok
20:58:23.304 -> ok

```
So i entered $$ then got this::
```
20:58:27.710 -> $0=20
20:58:27.710 -> $1=50
20:58:27.710 -> $2=0
20:58:27.710 -> $3=1
20:58:27.710 -> $4=0
20:58:27.710 -> $5=0
20:58:27.710 -> $6=0
20:58:27.710 -> $10=1
20:58:27.710 -> $11=0.050
20:58:27.710 -> $12=0.002
20:58:27.710 -> $13=0
20:58:27.710 -> $20=1
20:58:27.710 -> $21=1
20:58:27.710 -> $22=1
20:58:27.710 -> $23=7
20:58:27.710 -> $24=600.000
20:58:27.710 -> $25=3000.000
20:58:27.710 -> $26=100
20:58:27.710 -> $27=1.000
20:58:27.710 -> $30=1000
20:58:27.710 -> $31=0
20:58:27.710 -> $32=1
20:58:27.710 -> $33=350.000
20:58:27.710 -> $100=80.000
20:58:27.710 -> $101=80.000
20:58:27.710 -> $102=80.000
20:58:27.710 -> $110=60000.000
20:58:27.710 -> $111=60000.000
20:58:27.710 -> $112=60000.000
20:58:27.710 -> $120=2200.000
20:58:27.710 -> $121=1800.000
20:58:27.710 -> $122=2500.000
20:58:27.710 -> $130=160.000
20:58:27.710 -> $131=150.000
20:58:27.710 -> $132=1.000
20:58:27.710 -> ok
20:58:27.710 -> ok

```
However i don't really have a clue what I am doing.. lol...
When I open LaserGRBL i get this... 
[image]
If i go in LaserGRBL and go to Grbl -> Grbl Configuration.. I exported the following:
[LaserGRBL-Code.nc.txt](https://github.com/LaserWeb/LaserWeb4/files/3971230/LaserGRBL-Code.nc.txt)
---
Hope this helps!  not sure what to look for or send to the seller...

cprezzi, 2019-12-17:
Ok, it seems there is the correct welcome string, but your STM32 board doesn't send it automatically after serial connect.

Please save the following zip to the LaserWeb installation folder, then unzip it and run the create_env.bat file: [create_env.zip](https://github.com/LaserWeb/LaserWeb4/files/3972810/create_env.zip)

This creates an environment file (.env) that tells LaserWeb to reset grbl after serial connect. This should push grbl to send the welcome string.

Geoff-S, 2019-12-17:
Not sure if it's relevant in this case but I've been having a similar problem with a Protoneer Pi Hat. To solve it, I had to modify config.js and change the value for RESET_ON_CONNECT from 0 to 1. I do have RESET_ON_CONNECT=1 in a .env file but that didn't work.

cprezzi, 2019-12-17:
@Geoff-S This is something else, as you seem to use lw.comm-server, not the compiled release. With the realese, he dosen't have access to the config.js source file!

Stady234, 2019-12-17:
@cprezzi Worked perfectly!!  thank you... side question.. as this was the first test.. do i do the same for the pi server version?

@Geoff-S the batch file that @cprezzi sent basically does that... his procedure worked.. but thank you.. i will keep a loo out for this on any other installs.

Geoff-S, 2019-12-18:
@Stady234 That batch files creates the .env file which I already had. To make it work, I had to modify _config.js_ for the RESET_ON_CONNECT parameter to take effect. This on a Raspberry Pi thoguh.

Stady234, 2019-12-18:
@Geoff-S Ah... ok.. well that helps for my part2 question about how to do this on the pi!!!  ok... I will test soon and msg you if i have any issues...

cprezzi, 2019-12-18:
@Geoff-S Your case was special, as usualy the .env file solution also works for the pi.

@Stady234 Just use the wiki instructions for the installation on a pi:
https://github.com/LaserWeb/lw.comm-server/wiki/Manual-Installation-(RasPi)

Geoff-S, 2019-12-18:
@cprezzi Any suggestions as to why it might not have worked for me? I followed the instructions at https://github.com/LaserWeb/lw.comm-server/wiki/Manual-Installation-(RasPi) to the letter.
I have it working now so I'm happy but would like to know is it could be something I did incorrectly.

Stady234, 2020-01-19:
Ok, so i just finished the pi install... I can connect the desktop to the pi host... but when i do the machine connection it disconnects like before...  there is a "conf.env" in the /home/pi/lw.comm-server (from the installer, with "RESET_ON_CONNECT=1" in it... but to follow the instructions on the wiki 

> To do that, you have to create a file called ".env" (nothing in front of the dot!) in the installation folder of the server with the following content:
RESET_ON_CONNECT=1

I placed it in the same folder as there is no description of what else the "install" folder could be....  but it does not work...

Help please!

cprezzi, 2020-01-20:
The filename MUST be ".env" only (nothing in front of the dot)!

Stady234, 2020-01-20:
I did.... And as a reminder.... After we got the "Reset_on_Connect" part on the Desktop... I was able to connect just fine and control the laser.... This is to shift from PC to the PI for control...

[image]

cprezzi, 2020-01-20:
Are you sure you are using the correct serial port? On my setup it is "dev/tty_ACM0".

In some cases the ".env" file doesn't work. In this case you can try to change the presets in config.js directly.

Stady234, 2020-01-21:
I have been connecting to the only port it listed on the right status when the laser machine was connected...    
[image]

after it "connects" then 10 seconds later... 
[image]

But again... shifting this laser to the PC with a working ".env".. i have NO issues...
--
For the config.js -> what would i change? I don't understand...

cprezzi, 2020-01-22:
To activate reset on connect via config.js, change the line:
`config.resetOnConnect = process.env.RESET_ON_CONNECT || 0;`
to
`config.resetOnConnect = process.env.RESET_ON_CONNECT || 1;`

cprezzi, 2020-02-11:
@Stady234 Could you solve the issue with the config.js file?

Stady234, 2020-02-13:
Sorry for the delay;
That fixed it.... not sure why the original way you stated (the env file) did not work.  But i will be sure to document how to get to this point.

closed 2020-02-13 by Stady234

Nepherael, 2025-03-03:
> Ok, it seems there is the correct welcome string, but your STM32 board doesn't send it automatically after serial connect.
>
> Please save the following zip to the LaserWeb installation folder, then [...]

I know this is an old thread and I don't even know if this program is still kept up with. I've been using it for many years and my GRBL, getting very old, doesn't seem to be sending this welcome string. Saw you discuss this solution in a few threads but I could not find the install folder for the life of me and it isn't documented anywhere that I could find through Google or the documentation. I ended up running the installation file I downloaded back in 2018 in order to see if I could find where it was planning to install the program but instead of a standard install procedure it ran automatically which reinstalled laserweb and now every single profile and setting I had is gone

I barely got it operational again and finally remembered a way to check a file path while a program is open to find its root instead of just the shortcut it pulls from. 

I used this fix that creates the .env file and it seemed to work perfectly. I changed a few settings in the machine settings and the laser seems to be functional. Although my play/pause button doesn't seem to work right anymore (after hitting play it just says play still and have to click it twice more for a pause). Figured I'd post my experience with it in case this program is still worked on in any way

I found some profile and settings files that I seem to have saved back when I first got the program but when I go to the machine setting and try to choose them to upload them it just freezes the program amd corrupts the machine profiles file.