LaserWeb/LaserWeb4 #585: server.js breaks with serialport => 8.0.3

demigh0d, 2020-01-25:
As of version 8.0.3 (October 2019), serialport.list no longer supports a callback argument.

> 8.0.3
> BREAKING CHANGES
> drop callback argument on SerialPort.list() (#1943) (145b906]

```
Jan 23 01:43:59 lasercnc lw.comm-server[11498]: App connected! (id=0)
(node:11498) UnhandledPromiseRejectionWarning: TypeError: SerialPort.list no longer takes a callback and only returns a promise
lw.comm-server[11498]: at Function.SerialPort.list (/home/pi/lw.comm-server/node_modules/@serialport/stream/lib/index.js:651:11)
lw.comm-server[11498]: at Namespace. (/home/pi/lw.comm-server/server.js:162:16)
lw.comm-server[11498]: at Namespace.emit (events.js:305:20)
lw.comm-server[11498]: at Namespace.emit (/home/pi/lw.comm-server/node_modules/socket.io/lib/namespace.js:209:10)
lw.comm-server[11498]: at /home/pi/lw.com-mserver/node_modules/socket.io/lib/namespace.js:177:14
lw.comm-server[11498]: at processTicksAndRejections (internal/process/task_queues.js:79:11)
lw.comm-server[11498]: (node:11498)
```

demigh0d, 2020-01-26:
The source at github.com/LaserWeb/LaserWeb4-Binaries does not list a version requirement for serialport in package.json.

cprezzi, 2020-01-28:
This relates to lw.comm-server (which is a separate repository) and should be posted there. 
We rely on Serialport 6.2.2 and NodeJS 10.x!

closed 2020-01-28 by cprezzi