LaserWeb/LaserWeb4 #586: job simulation whith relative cordinates (G90)

ithinkido, 2020-01-30:
When I am running a job from Gcode using relative co-ordinates , I get a perfect rendering on screen of the job, but the actual blue laser pointer is not moving on the rendering when the job is running, rather the whole rendering + blue dot ( laser) is moving around the screen.  Is there a way to see the simulated position of the laser durring the job ?

[Laserweb-Workspace.zip](https://github.com/LaserWeb/LaserWeb4/files/4136763/Laserweb-Workspace.zip)

cprezzi, 2020-01-31:
Do you use a GRBL baser controller? 
If so, do you have conigured $10=0 as recommended?

ithinkido, 2020-01-31:
Nope, I was not using $10=0. That was my problem. Thanks.

closed 2020-01-31 by ithinkido