LaserWeb/LaserWeb4 #588: Font not visible

Dane809, 2020-02-28:
I draw something in Inkscape with various fonts.
One font is visible, the other is not visible. 
All fonts are installed on Windows 10 in the font folder.
What can I do? 
I won´t to export the Inkscape file as PNG

pedjas, 2020-02-29:
Have you tried to convert text to paths?

Dane809, 2020-03-01:
Yes but not what i want. I have a workaround. Export text as png and then import in the svg.

cprezzi, 2020-03-02:
Normal Fonts need to be converted to path before exporting to SVG, otherwise LW can not read them. In LW you can then decide if you want to burn the outline or do a raster fill.

What your workaround does is rasterizing the text as png and then embed that raster data into the svg. This also works and is a good solution if you want to do raster engrave and vector cut in the same job.

closed 2020-03-02 by cprezzi