LaserWeb/LaserWeb4 #589: SVG objects ar enot supported

pedjas, 2020-02-29:
I am novice LaserWeb user. I use Inkscape to create SVG which then I import into LaserWeb to prepare gcode for laser engraving.

When I load svg into Laserweb it recognizes layers and objects in SVG and I can prepare operations based on them.

BUT if object I use is not converted from object to path in Inkscape before SVG is saved, LaserWeb does not create gcode for those objects. I have to convert objects to path before saving.

It would be great if it can crete GCODE from objects too. Reason: if I do some graphics that contains text and I want to frequently repeat engraving but just changing text, need to convert text to path complicates workflow a lot.

If that is not possible, then it should warn user that it is not possible and he has to convert to paths. As novice I wasted lot of time to understand why gcode is not generated. Warning would clear confusion.

cprezzi, 2020-03-01:
The library used in LW for parsing SVG doesn't support obects, but it's just one keypress in Inkscape to convert objects to path. I belief that's easy enough.

pedjas, 2020-03-01:
Yes it is easy to do, but if you have to repeat the same job but change text in it - it gets complicated - you have to keep version with text and then after text update convert to path and save as different file, and you can easily make mistake and destroy original.

But if LaserWeb cannot do object that is it, we have to live with it.

However, I proposed, it should warn user about SVG objects and do not allow user to use object to create operations. As novice, I spend lot of time trying to figure out why it does not process SVG I load.

cprezzi, 2020-03-02:
Unfortunately I cannot change the SVG library, but what you describe is the the typical workflow for every grafic designer. They typically do the convert to path as the last operation before delivering the data to the printer and call the file ..._path.svg

closed 2020-03-02 by cprezzi

pedjas, 2020-03-03:
Please, just warn user if there is object in SVG that cannot be processed as it is not path. That is all what is needed.