LaserWeb/LaserWeb4 #590: LaserWeb4 development, Linux, the universe and everything

janschiefer, 2020-04-11:
LaserWeb4 *seems* to have dropped support for Linux a while ago. This is a major problem for many users, since LaserWeb is the only "serious" laser engraving solution for GRBL users out there.

However I cannot really comprehend this, since I am able to run the program without major issues on Linux, using the current development branch dev-es6 from 12th April of 2020.

Just run:
npm run installdev (installs all dependcies)
npm start (compiles and starts the server +  app)

...and you are ready to go.

The only thing I have to figure out yet, is how to build an proper AppImage or .deb from the current development branch...

Any comments or suggestions?

cprezzi, 2020-04-12:
You are very welcome to contribute and care about the AppImage! Unfortunately most devs have left the project and I don't know much about linux.

janschiefer, 2020-04-13:
Ok, why not.

Unfortunately I don´t have much experience with the AppImage ecosystem (and also I am a physician in COVID-19 stress). However, Linux and CNC is my hobby for relaxation.

Anyone know a good "quick and dirty" tutorial for packaging nodejs-Applications into .AppImage or .deb?

cprezzi, 2020-04-13:
A good starting point should be 
https://docs.appimage.org/packaging-guide/index.html

cprezzi, 2020-04-13:
You should also check https://github.com/LaserWeb/lw.comm-server, especially the scripts in package.json and the docker files.

mathematicalmichael, 2020-04-16:
Might I suggest packaging it into a docker container? Would probably be much easier than packaging it as a deb.

janschiefer, 2020-04-16:
Trying to build the AppImage.
I'm having the LaserWeb4 and lw.comm-server in parallel directories.

Seems I'm missing some resource files...
---
(...)
packaging for linux x64 using electron 2.0.18 to dist/linux-unpacked
Unhandled rejection Error: Exit code: 1. Command failed: gm convert -size 48x48 /tmp/electron-builder-N3DabI/0-2-linux.iconset/icon_128x128x32.png -resize 48x48 /tmp/electron-builder-N3DabI/0-2-linux.iconset/icon_48x48x32.png
gm convert: Unable to open file (/tmp/electron-builder-N3DabI/0-2-linux.iconset/icon_128x128x32.png) [Datei oder Verzeichnis nicht gefunden].

gm convert: Unable to open file (/tmp/electron-builder-N3DabI/0-2-linux.iconset/icon_128x128x32.png) [File not found].
(...)
---

Am I missing a git submodule or repository with the image/icon resource-files...?

janschiefer, 2020-04-17:
Ok.
I have created working builds of the latest git version of LaserWeb4 for Linux x64 and Windows x64 with a little hacking: https://drive.google.com/open?id=1oGcQKB0K32n3B7irCJjCIg7AIyqRk-hz

- I removed all references to any icon files from package.json in ls.comm-server, so the default electron-builder icon is used.

- I created a symlink to "libreadline.so.6" (this version is not included in Ubuntu 20.04 Beta) from my system libreadline.so.

You have to manually open 127.0.0.1:8000 in your browser to open the GUI in this build.

janschiefer, 2020-04-20:
I'll try to create a patch for the build problems in the next few days.

How can I submit the patch?

deanforbes, 2020-04-21:
will this be able to run on a raspberry pie ? would it be possible to test it on that ?

cprezzi, 2020-04-21:
There is a installation guide fro rasperry pi on the wiki! 
https://github.com/LaserWeb/LaserWeb4/wiki/1.4-Installation-on-RaspberryPi

danielkucera, 2020-05-08:
What about using Docker? #595

zettrik, 2020-08-04:
Thank you Daniel, it's ist working fine with Docker - also on raspberry pi.

joesantosio, 2022-09-25:
I've just created a pull request to fix the dockerfile https://github.com/LaserWeb/LaserWeb4/pull/647
After merge, we could setup it on dockerhub for example and easily run it with volume linkage (for data persistence) even on a homelab setup. If you just care about the gcode generation, you could also just run it on a small server instance on linode / digitalocean...

The package server is indeed flawed. It seems node-gyp isn't able to have access to the system ports and fails, even with root permissions.

cprezzi, 2022-09-27:
@joesantosio I've merged your pr #647
Could you set it up on dockerhub? I have no clue about docker ;) 

About the system ports. Could this be caused by missing/incompatible libusb and libudev, as @QuirkyCort wrote in issue https://github.com/LaserWeb/LaserWeb4/issues/648?

joesantosio, 2022-10-03:
@cprezzi so I've created the docker image under my user. We can change it for a `laserweb` account but for that you will need to create the laserweb dockerhub account, create the access token and add it to the secrets here on the repository as `DOCKERHUB_TOKEN` and the username as `DOCKERHUB_USERNAME`. For now, I am running the CI on my fork on the branch `prod` which is intended to be the release branch.

I set up a PR to modify the `README.md` as per the changes ( https://github.com/LaserWeb/LaserWeb4/pull/650 ).

Did anyone test a higher version of node? 10 is quite low and it should be bumped.

harlock999, 2022-10-03:
Right now i'm running on @easytarget fork that has newer nodejs version with good success. My understanding is it was quite some work. Idea is eventually pull easytarget version back into the main LaserWeb4.

@easytarget : Any idea when you'll release you'll issue a pull request?

joesantosio, 2022-10-03:
@cprezzi by the way, I don't have other way to communicate with you, I am starting the work on a decoupled API in Go.

cprezzi, 2022-10-03:
> @cprezzi so I've created the docker image under my user. We can change it for a `laserweb` account but for that you will need to create the laserweb dockerhub account, create the access token and [...]

The actual servrer uses node 12 and it also seems to be working with node 18 (according to https://github.com/LaserWeb/LaserWeb4/issues/648).

cprezzi, 2022-10-03:
Why do you think we need a go version?
The node js version is very versatile and I can't support go.

joesantosio, 2022-10-04:
First of all, I don't think we need per se a Go version and take all of the next rant as an opinion and nothing more than that. 

In my opinion, we do need a fully decoupled backend from the frontend. The communication is already in place for this to happen and it makes it easier and more flexible to have that separation. You can have the frontend wherever you like running wherever you like and the backend, could run even on a smaller microcontroller (theoretically). Being decoupled also means that the server doesn't depend so much on the frontend and vice-versa which means that you can make better changes on either stack, you can assign people for more frontend or more backend and things tend to evolve a bit faster in that environment because the modularization helps to understand the development thinking.

Since, in my opinion, this is a good change, better to do it "right" this time around (not saying it is "bad", just that it could be "better".

Architecture is one thing where I think things could be improved. A pattern similar to DDD for example for better separation of concerns and modularization in general. 
The language is other topic where I think it could be improved. JavaScript by nature is flawed. Its dynamic abilities is also one of the cancers of the language. It opens to insecurity in the code specially while developing bugs arise more often and you are never really in a position that you can confidently state that your code works. The "node_modules" folder hell and dependencies in general is other cancer. There are a million of horror stories and I bet you have yours.
With that in mind, in my opinion, there are 2 contenders for this spot. TypeScript and Go. I work everyday with the 2 by the way. 
Both are compiled (even though TypeScript can also be a transpilation layer to JavaScript), both have typing and in general both are better development experience. 
TypeScript, as kind of a superset of JavaScript, also inherits its flaws. The dependency hell, you can circumvent the safety, if used as a transpilation layer its dynamic nature that might not work somewhere else or even if compiled might be a bit of a challenge somehow with some dependencies, node in general is harder to deploy. 
Go is strongly typed, compiles to a single binary file that doesn't need anything else to run, the language itself is quite easy to understand and learn in 1 week, no magic involved, it is a more robust language, it doesn't need 2gb of memory to run, 100mb is just fine or even less, I prefer Go C binds, a testing framework included, a good standard library with probably most of what you need, the Go community doesn't do dependency packages just because (there are no 3 line packages like in the node world and for some reason are used everywhere?!) and quite a bunch of smart people, easy compilation, good tooling...

I just find Go to be a better overall language and experience for the backend side of things.
Sorry about the long explanation. Tried my best.

joesantosio, 2022-10-04:
This all said, I am just dwelling with that idea of Go. I do think, again in my opinion, that at least TypeScript should be used but probably the first steps would be to setup the architecture, decouple, modularize and unit tests so it stays stable regardless of the language.

cprezzi, 2022-10-04:
It is very easy to make a decouppled backend version. We just have to drop the embedded website and http server or even better replace the embedded website with a generic "machine control" / "gcode sender" frontend, but I would not drop the option to have the LaserWeb (or any other) frontend embedded, because it is very convenient for the user, not needing to install any frontend app.

Actually, it would already be possible to embed any other frontend by just replacing the files in the app folder, as long as the frontend uses our socket api.

joesantosio, 2022-10-04:
Yes I can see the advantage of having the same package with all. I think there are ways though to still maintain the all-in-one package with both decoupled.

Well... it isn't really that easy to maintain as is :p but I understand you. I was thinking on just making a simple PR proposition on how I see this working (foregoing the Go and language change at the first instance) but since someone else is already working on a reviewed version, starting from there makes perfect sense.