LaserWeb/LaserWeb4 #592: White screen / blank screen when I start the program

vaiosx, 2020-04-14:
Hello

When I start laserweb4 I get white screen and nothing more.

1) It's first installation
2) I tried different versions, from 2018 to 2020

**Configuration:**
- Intel Atom 2700 / 2GB / SSD
- Windows 7, 32bit
- No internet on PC, no antivirus or firewalls

cprezzi, 2020-04-14:
- Do not use the newest version (as stated on the release page)! Use version 4.0.996-134!
- If you already have installed the -136 version, uninstll it and install version -134.
- If that still does not work, it might be your PC. Intel Atom might be missing something that LW4 needs (like GDI Drivers).

Have you tried it with a different PC?

vaiosx, 2020-04-14:
Thank you for answer.

136 - I not tried newest version.
Tried - 4.0.994-130, 4.0.996-130, 4.0.996-134 - all show white screen.
Also, I connect the internet, and try with it, same result.
On other computer work fine.

Might be need some drivers from microsoft visual C++ package or something like this.

mfadel85, 2020-04-22:
The version LaserWeb.Setup.4.0.996-136.exe worked on one of ours PCs, with Windows 10, i5-8265U.

cprezzi, 2020-04-22:
@mfadel85 Thank you for the info. I have just made that version a few days ago.

vaiosx, 2020-04-22:
My problem is not resolved, because there no way to know which library need for working LaserWeb on Intel Atom 32bit PC-s with Windows7.

cprezzi, 2020-04-22:
I'm not sure if it even can be run on Intel Atom, sorry.

You can try to install chrome browser and open the following URL:
https://laserweb.github.io/LaserWeb4/

If that also does not work, then this Atom PC might be incompatible.

closed 2020-08-06 by cprezzi