LaserWeb/LaserWeb4 #593: Language

arthurwolf, 2020-04-15:
Just a general remark.

Comments such as this one on the grbl-lpc page: « We have ported the ingenious GRBL 1.1 to LPC1769 boards (like Smoothieboard, Cohesion3D, Azteeg X5, MKS SBASE) because Smoothieware has no decent realtime streaming protocol, which is needed for fast raster engraving with realtime control. »

I think that's a very un-necessarily conflictual way of putting it, and you don't need to be that confrontational. We 100% disagree that this is the only way to achieve fast engraving with realtime control. I'm not here to debate that ( and am too busy to attempt having this conversation again ), I'm just saying even if you are completely certain you are right on this, even if let's say I'm wrong about this, there are ways of saying this that are less confrontational, and we'd be thankful if you used those ways of communicating about the issue rather than the way you're using right now.

You are 1000% in your right to talk about this any way you want, this is just me telling you what my preference is in terms of how I'd like you guys to talk about it. Feel free to completely ignore me, this is just me telling you how to be friendlier if you want to be. Which I think in the long run would benefenit everybody ( being friendlier that is ).

Cheers.

cprezzi, 2020-04-15:
Hi Arthur

I can understand your point and I must admit that this part has been added long time ago when everything escalated and I found on http://smoothieware.org/software that LaserWeb is stated as "not recommended to use as a streamer of rasters as they do not support streaming the smoothie way".

I would love if we both could remove these "side blows".

cprezzi, 2020-04-15:
Where did you find that "confrontational" comment?

On the LW4 Github REDME.md I have replaced it some time ago with something more neutral:

> If fast raster engraving is important for you, we recommend replacing Smoothieware with grbl-LPC (https://github.com/cprezzi/grbl-LPC) which also runs on the LPC1769 based boards and performs much faster for laser raster applications.

arthurwolf, 2020-04-15:
Somebody sent me this link:
https://github.com/LaserWeb/LaserWeb4/wiki/2.2-Initial-Configuration---GRBL-LPC-1.1
trying to argue that Smoothie's "ping pong" protocol sucks :)

About the comment on the smoothie software page, I'm not actually clear
right now on how well Smoothie streaming is supported in laserweb / what is
the performance when streaming raster, compared to some we know work well
like lasergrbl or smoopi. If you tell me it works like the ones we know
work optimally, I'll edit the comment for sure. If there's no change to the
performance, but you have an issue with how the comment is worded, I can
definitely work on that too.

Cheers.

cprezzi, 2020-04-16:
Thank you for the link. I have changed the text there.

The streaming in LaserWeb works with defined packet bursts, so we still have the chance to send a pause/stop or feed change while executing the file with not too much of a delay. As a result, we will probably not reach the same speed on Smoothieware as apps that use the Smoothie Streaming. Sending the whole file at once whithout a chance to interrupt (via software) during execution just doesn't seem right to me (especially for CNC Mills).

Cheers.

closed 2020-04-20 by cprezzi