LaserWeb/LaserWeb4 #597: Loading SVG from memroy(generated in the web application)

mfadel85, 2020-05-21:
Hello everyone,
I want to generate an svg file inside the application and load it direclty to generate gcode, any clues how to start?

Many thanks

cprezzi, 2020-08-06:
The best would probably be to add a new tab "creator" after the "files" tab, in which you load the XML, create the SVG and add the SVG to the workspace. This has to be done the same way like the other tabs (with React & Redux).

mfadel85, 2020-08-07:
Thank you , I implemented it, I used [MakerJS](https://maker.js.org/) to generate svg from text,
 I just have to clean and organize it.

easytarget, 2020-08-17:
I'm intrigued: Being able to generate Text, (and outlines/borders? consider this a feature request), directly in LW would be awesome. One less reason to bang my head on the desk while using Inkscape.