LaserWeb/LaserWeb4 #599: Reset Settings (e.g. Machine Profiles) to default

shachar-limor, 2020-06-25:
Hi,
I'd like to reset the Settings section to it's default - would like to do a little cleanup.
I thought I'd find a solution to this by uninstalling and reinstalling (does not work).
Also I tried to find the laserweb-settings.json and remove it - but I could not find the path..

Please advise,
Shachar

NikhajZzz, 2020-06-29:
Hello! 
My app freezed on OSX because of wrong grid settings. Now it is unusable. I need to reset my app settings too. Reinstall wont work. LaserWeb app is not storing any settings in the user/library/preferences/ as usual the OSX apps do. The settings are stored somewhere in java or own database probably. How to reset the app settings???? HELP US PLEASE!

shachar-limor, 2020-06-29:
I did not mention before, I am using Win64.

Dane809, 2020-08-12:
Please try to clean the browser cache. I think the custom settings are stored there.

cprezzi, 2020-08-12:
Yes, the settings are stored in the browser storage, but the compiled version of LaserWeb is an Electron App which has an embedded chromium browser, so you don't have access to the browser frontend to clean the cache.

The embeded browser stores it's data in these locations:

```
Windows:
C:\Users\<user>\AppData\Roaming\lw.comm-server

Linux:
/home/<user>/.config/lw.comm-server

OS X:
/Users/<user>/Library/Application Support/lw.comm-server

!! REPLACE <user> WITH YOUR USERNAME !!

```
To clean all data you just have to delete the lw.comm-server folder.

NikhajZzz, 2020-08-13:
Hi, thank U for the answer, i have finally found my way out. Deleted my App support files and the reset was done. Finally i have imported my good settings file and all is working like a charm.

closed 2020-08-14 by cprezzi

shachar-limor, 2020-08-15:
Thanks all for your responses and help :-)