LaserWeb/LaserWeb4 #601: problem when using "mill pocket" (sticks at 50% indefinitely)

alanesq, 2020-07-13:
I have built myself a CNC router and would love to use LaserWeb with it as it is an amazing bit of software but I have tried everything I can think of but am unable to get "mill pocket" to work with most of my files.

Unless the SVG file is something very basic I find it just gets to 50% and never gets any further.  
If I set the cutter to a very large size (e.g. 10mm) then this will sometimes get it to work on some basic shapes (my usual cutter size is 3.175), changing the step-over and segment also can help in some cases but I find most of my projects I am unable to get it to work.  I usually end up having to go back to JSCut in most cases :-(

After some experimenting I have found that if I draw a square box (I use Inkscape) this works no problem in seconds, if I then stretch one of the sides out in an arc this also works ok but if I reverse the stretch direction so that one of the sides arcs inwards then this causes the problem and it fails to complete. 
I have attached the SVG files I am using here: [SVG-files.zip](https://github.com/LaserWeb/LaserWeb4/files/4913277/SVG-files.zip)

[Laserweb-Workspace.txt](https://github.com/LaserWeb/LaserWeb4/files/4913246/Laserweb-Workspace.txt)

celianvdb, 2020-07-22:
Hi,
I'm having exactly the same problem. I noticed that the tool size does have a big effect.
Last tried I made 
3 mm : ok ( couple of seconds )
2 mm : ok ( couple of seconds )
1.1mm : ok ( couple of seconds )
1 mm : Didn't finished after 10 minutes.

I can help by anyway don't hesite !

alanesq, 2020-08-20:
Something which may give a glue as to what the problem is?

I created some basic text 100mm x 30mm (using Inkscape) and selecting the four letters individually in LaserWeb created a mill pocket operation, this takes around 12 seconds to calculate - no problem there.
But, 
[text-demo.zip](https://github.com/LaserWeb/LaserWeb4/files/5101201/text-demo.zip)
If I then combine these four letters in to a single path (in Inkscape by selecting PATH, UNION) so that I can then select the whole word in LaserWeb with a single click, when I then try to create a pocket operation it is still stuck at at 50% after 5 minutes...

Note: I have attached the files I used

cprezzi, 2020-08-20:
@alanesq Instad of combining the objects, you should just group them (CTRL+G).

alanesq, 2020-08-20:
> @alanesq Instad of combining the objects, you should just group them (CTRL+G). Even with separate objects you should be able to select the whole file to create an operation for all.

Hi,
I tried this but you still have to select each letter individually in LaserWeb (not a problem with this 4 letter word but can be a pain with a lot of text if you have to select the parts you want an operation to act on).