LaserWeb/LaserWeb4 #602: Electron-Packager: syntax error: cannot use import statement outside a module......

mfadel85, 2020-08-06:
Thank you everyone,
I am trying to generate an executable file in windows using electron-packager, no error happened but when I run the generated execuatble file in windows 10 it shows 
syntax error: cannot use import statement outside a module......
How can I solve this problem? Or how can I force electron-packager to use node 10, will it work?
I attached a screenshot of the message error when running the generated windows application
[image]

Best Regards

cprezzi, 2020-08-06:
Did you follow the "How to compile" giude in the wiki?
https://github.com/LaserWeb/LaserWeb4/wiki/How-to-compile

mfadel85, 2020-08-07:
> Did you follow the "How to compile" giude in the wiki?
> https://github.com/LaserWeb/LaserWeb4/wiki/How-to-compile

it worked, thanks,

closed 2020-08-07 by mfadel85