LaserWeb/LaserWeb4 #612: No Supported Firmware - STM

choadalish, 2020-09-04:
Hello,
I've been trying to get the Ortur Laser Master 2 to work with LaserWeb 4 but receiving the "No supported firmware" message. I worked through the issue "No supported firmware detected. - GRBL1.1f #584" but my .env file does not seem to be affecting LaserWeb and I'm wondering if I have it in the right place.
  
**OS**: Currently running Windows 10
**.env File location:** C:\Users\chkummro\AppData\Local\Programs\lw.comm-server\.env
**Ortur Board:** STM32 MCU

When connecting with LaserGRBL I receive:
**_Ortur Laser Master 2 Ready!
OLF 137.
Grbl 1.1f ['$' for help]_**

mw75, 2020-11-24:
I don't know your runtime (Node on Windows), but i have the same Device running with resetOnConnect set to one. I just have the setting in my config.js, but you can also set it by RESET_ON_CONNECT in the .env-File.
config.resetOnConnect = process.env.RESET_ON_CONNECT || 1;

cprezzi, 2020-11-25:
@choadalish The ENV file must not have anything in front of the dot. The filename should be ".env" only. Did you eventually forgot the backslash \ between lw.comm-server and .env?

cprezzi, 2020-11-25:
@mw75 Please be aware that the normal usecase with the installer exe can only be configured by the .env file. The config.js file is only avalable when you run the source code direclty with nodejs.

SKucharski, 2021-02-09:
Where is config.js in ubuntu? I get LaserWeb Appimage. My machine is ortur laser master 2.

cprezzi, 2021-02-10:
> 
> 
> Where is config.js in ubuntu? I get LaserWeb Appimage. My machine is ortur laser master 2.

As I wrote in the last post, config.js is only available if you run from scource code with nodejs. Follow the Raspbery Pi installation instructions.

closed 2021-02-10 by cprezzi