LaserWeb/LaserWeb4 #613: install serialport errors on raspi

a7hybnj2, 2020-09-29:
`uname -a` - `Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux`
`npm -v` - `6.14.6`
`node -v` - `v10.22.1`

Installing via this [guide](https://laserweb.yurl.ch/documentation/installation/36-install-raspberry-pi)

Getting the following errors from this `sudo npm install serialport --unsafe-perm --build-from-source` command.
```
found 42 vulnerabilities (37 low, 2 moderate, 3 high)
  run `npm audit fix` to fix them, or `npm audit` for details                                                      
pi@raspberrypi:~/lw.comm-server $ npm audit
npm ERR! code ELOCKVERIFY
npm ERR! Errors were found in your package-lock.json, run  npm install  to fix them.
npm ERR!     Missing: hdiutil@0.0.6

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-29T00_09_30_255Z-debug.log
pi@raspberrypi:~/lw.comm-server $ npm audit fix
npm ERR! code ELOCKVERIFY
npm ERR! Errors were found in your package-lock.json, run  npm install  to fix them.
npm ERR!     Missing: hdiutil@0.0.6

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-29T00_09_56_322Z-debug.log
```

I tired to manually install the package but can't see to find it or figure it out.

cprezzi, 2020-09-29:
Did you try to run `sudo npm install` as the error message is sugesting?

a7hybnj2, 2020-09-29:
Yes. I have now run `sudo npm install`, `sudo npm audit fix`, and `sudo npm audit fix --force`. I hope I am getting close to all the packages being installed.

cprezzi, 2020-09-30:
Do NOT run the `audit fix` commands as this will update the used libraries to incompatible new versions!

a7hybnj2, 2020-09-30:
😬 guess the best thing at this point would be to start over and not try to fix any errors.

closed 2021-02-10 by cprezzi