LaserWeb/LaserWeb4 #614: "Check Size Power" & "Tool Test Power" values are too High

friiidge2, 2020-10-27:
These 2 Functions do not allow to insert a floating point number under 1.
With 15W Laser If i use 1 as "Check Size Power" or "Tool Test Power", the Laser burn the wood

Is it possible enable the use of floating point number

Regards

cprezzi, 2020-10-27:
This sounds like a problem with your firmware configuration. On GRBL the default PWM range is 0-1000, which means 1 is 0.1%.
What firmware are you using?

friiidge, 2020-10-27:
Hello,

You're right, i want to introduce 0.1% as value.
But LaserWeb4 do not authorize the value. When you save it, it automaticaly come back to 1
[image]

My firmware is GRBL 1.1H with Following settings
$30 = 1000

But your calculation is false :-)
1% of 1000 = 10 (PWM and not 1 PWM)

for this reason my Laser burn material.
I have to reduce $30 = 100
In this case, with a settings of 1% in "Check Size Power" & "Tool Test Power" I can use a real value of 1 PWM
Then i have to change the $30 =1000 

Do you think possible to authorize floating point numbers for these 2 values?

cprezzi, 2020-12-01:
On Arduino Uno or Nano, the real PWM values are only 8 bit (0-255). Because of memory and speed limition, grbl internally operates with integer value, not floating point. S value are scaled according to $30. With $30=1000, 1% equals to 10, which is scaled (divided by 1000 and multiplied by 255) to a PWM value of 2 or 3 (depending on rounding). Such a value should not deliver enough power to burn anything. If it does, you have a problem with your laser (probably not a TTL PWM compatible laser driver).