LaserWeb/LaserWeb4 #615: DXF 2000 and later support

AleyRobotics, 2020-10-30:
Please add DXF 2000 or later file format support

easytarget, 2022-01-12:
A quick, belated, note that this is one of the issues addressed in my recent work; you can see a preview here: https://easytarget-org.github.io/lw-mods-preview/ , I hope this will make it to the main code base in the near future.

easytarget, 2022-02-08:
It will be done when it is done; in the meantime you can try the preview listed above. It produces viable gcode that can be saved and then imported & run via web UI's, ugs, etc..

harlock999, 2022-02-09:
Awesome, thanks for this. I'm using Debian Bullseye by the way on a Amazon bought Quotom Q355g4 with 8gb ram and SSD. At the price I got there wasn't much point to fiddle with RPi. Will look into this soon.

cprezzi, 2022-02-09:
@harlock999 I have implemented support for XHC HB04 into lw.comm.server several years ago (including DRO display). Check out [MPG branch](https://github.com/LaserWeb/lw.comm-server/tree/MPG) of lw.comm-server, if your courious how I did it. ;)

harlock999, 2022-02-09:
Thanks Claudio, I'll definitely have a look. The HB04 is a bit different than the HB04**B** so the bitmapping is different. But for communication level I will check because right now I had to hack the node-usb in some not-very-elegant way: some functions were dumbing down USB comms too much and I didn't had access to the special code the HB04B was requiring. I tried also to put the MPG at the web browser client level like the gamepad but I couldn't get much done due to some precompiler requirement that the web browser wouldn't do.

cprezzi, 2022-02-09:
I implemented it server side, because the machine is also connected to the server and this is where the MPG is used.
If you like it, I could probably pull over my additions to the actual master branch.

harlock999, 2022-02-09:
Let me have a look first.

easytarget, 2022-02-09:
@harlock999 
Both myself and @cprezzi  hang out on MakerForums (see all the links in readme's etc), there is a development forum there and you can seamlessly login via github authentication.
I'll be making some posts there summarising my next steps asap.

I get the need to get this all commited asap, and free the dev-es6 branch for other developers.

AleyRobotics, 2023-08-29:
Are any update on this?

harlock999, 2023-08-29:
I've been using @easytarget fork for the front end over a year now, albeit I'm saving into R12 format. I'm using my server for HB04B MPG support and it's quite reliable for what I do.