LaserWeb/LaserWeb4 #617: Step and repeat function missing + rotated table (-Y, -X) and homing help + newer OSX version?

NikhajZzz, 2020-12-04:
OSX, Laserweb 4.0.966

1. is there a newer version for OSX than 4.0.996? 4.0.998-136 dmg file?

2. Missing step&repeat a lot. We cut boxes from paper for testing the new constructions.

3. Our laser has a turned table, our home with sensors is upper right corner. This is because we need to place larger sheets into the machine. I cannot solve to turn the table in laserweb settings, using $3=3 sets the home position wrong. Machine bounds problems. I have to use negative coordinates now, but i want to turn it into positive, so -250mm should be translated into 250mm - same as the table with bottom left homing. Please help me to solve this! 
 
[image]
[image]

bridgeofstraws, 2022-08-27:
I wonder if the homing direction config parameters would help you?
https://3dtechworks.ca/2020/02/12/grbl-settings-and-calibration/

$22=1 (homing cycle, bool) – enable/disable homing cycle
**$23=3 (homing dir. invert mask: 00000011)** – sets direction of travel for homing cycle for each axis. Last 3 bits set direction for ZYX respectively
$24=200.000 (homing feed, mm/min) – speed of movement for last part of homing cycle
$25=1000.000 (homing seek, mm/min) – speed of movement for first part of homing cycle
$26=50 (homing debounce, msec) – debounce time in milliseconds for homing switches
$27=1.0 (homing pull-off, mm) – amount of pullback when homing routine hits a switch. Ensures that switches are not still activated after homing cycle if limit switches are enabled.

cprezzi, 2022-08-30:
The newer releases ar in the Binaries Repo: under https://github.com/LaserWeb/LaserWeb4-Binaries/releases

closed 2024-05-31 by cprezzi