LaserWeb/LaserWeb4 #623: Marlin/Linux No supported firmware detected. Closing port

jimhiggs, 2021-01-28:
I have just installed on debian 10 linux from the latest appimage on the releases page. My CNC is Marlin 2 on a ramps board. When trying to connect I am getting the no firmware message. I have researched online and read about changes to a .env file. Having used an appimage file I don't see any files except the appimage file itself. 
I tried adding a .env file in the same folder as the executale appimage but believe it is ignoring that. 
Any idea?

cprezzi, 2021-01-28:
I don't know anything about appimages and the developer that was responsible for the appimage generation left the project.
You could try to manually install LaserWeb by following the Raspbery Pi installation guide on
https://laserweb.yurl.ch/documentation/installation/36-install-raspberry-pi

jimhiggs, 2021-01-29:
Ok Thanks, will give it a try.

jimhiggs, 2021-01-29:
Working nicely now thanks

closed 2021-01-29 by jimhiggs