LaserWeb/LaserWeb4 #625: Feature Request For Laser Cutting

dcwalmsley, 2021-06-15:
I have a NEJE 20W laser installed on a quick release for my Ender-3V2 printer.  

I started testing and using your LaserWeb application but couldn't find a feature to allow for the laser to drop in height by millimeters for cutting thick material such as birch or plywood to name a couple.

The laser burns away layers at the focal point but becomes less effective as it is burning normally at the surface layer vice adjusting each pass to a new focal depth.

Please consider adding the ability to allow software to automatically adjust depth per pass based on Material thickness, Laser speed and Number of passes.  This should be made to allow operator to make manual adjustments.

This feature could be similar in nature to how CNC mill away layers if this is adoptable to lasers.

Thanks,
Doug W

cprezzi, 2021-06-25:
Did you try to use a "Mill Cut" operation instead of "Laser Cut"? In the RPM field you can just enter the desired PWM value.

easytarget, 2022-01-12:
Support for Z axis lowering during lasercutting is already in the code; you should see a 'start height' and 'pass depth' in the laser cut operations. If not, make sure you have `Z Stage Support` enabled under the Machine settings, you will not see anhy Z/height settings in the operations panel unless this is checked.

harlock999, 2022-01-31:
I had the same as dcwalmsley. My laser is quite powerful and I need to "drill" the metal with the laser before I can start cutting with a kerf. This is not a multi-pass depth thing. Everytime there's a new path, the laser head needs to back up using a G0 then lower with a G1 move. What I ended up doing is in the TOOL ON Gcode setting is I have the following:

```
;Piercing
G0 Z5 F2000 ; back up 5mm
G1 Z0 $INTENSITY F2000 ; lower and pierce

```
By doing it like this I save the optics a bit as there's more room for molten metal to fly away. This is obviously a crude patch and works only for laser cutting with a workpiece at Z +0.000. Multiple pass setup and raster-type operations may require a different machine definition or disable that piece of code. I'm mainly using Autocad DXFs and it highlighted some other issues with the GCode generator. If I'm using lines to "build" my path of say a square, then the original R12 DXF processor is building 4 paths for each side of the square, making a piercing starburst-looking defect at each corner. With a polyline, I can have piercing done within a small line of 3-4mm inside the cutout shape, then move to a corner and complete the square. This moves the starbust defect inside the waste material. However, the GCode is so smart that it optimizes the next path using closest neighbor (I think?). I saw also an old open issue about Travelling salesman problem (TSP) optimizer, well in my case I'm seeking the opposite, being able to disable any kind of optimizer and simply start the GCode at the first vertex encountered, which would be my piercing location.

In the best of worlds we'd have a piercing function toggle within the lasercut GCode generation profile along with a optimizer selection (off, nearest neighbor, TSP, etc).

Looking forward for easytarget improvements!