LaserWeb/LaserWeb4 #627: Connection Issues GRBL ESP32 Serial

ithinkido, 2021-06-26:
I am having a hard time connecting to a fresh install of GRBL ESP32 using a serial connection. I am running the TestDrive configuration  to keep testing as standard as possible.
I have seen the  issue about adding reconect to the .env file and I have doen this, as well as increase the firmeware detection time to 10 sec. Nevertheless, there is still no connection to the machine, only an error with "firmware not detected, closing port "
I have attached the enamed  the .env(.txt) file that I have placed in 

C:\Users\USER\AppData\Local\Programs\lw.comm-server

[.env.txt](https://github.com/LaserWeb/LaserWeb4/files/6720680/default.env.txt)

[image]
Please can you tell me if there is anything else needed to use this firmware with Laserweb

cprezzi, 2021-06-27:
I never tested GRBL ESP32 with LaserWeb. Can you please post a link to the firmware you are using?

The ENV File you posted has .txt at the end. This is wrong. 
The file must only have .env as the filename, with no additional ending. Unfortunately the default windows file explorer settings hide the ending of known types like .txt. You first have to disable this hiding to see and change it.

ithinkido, 2021-06-27:
Sure, this is the firmware I am using 
https://github.com/bdring/Grbl_Esp32

I just saw that I had a typo in my first message that may have made it difficult to understand. I have corrected that

  The file I attached was called .evn.txt . I re-named it so that I could uplaod it to github as I was not able to upload a .env file extension. The original file that is saved in my C:\Users\USER\AppData\Local\Programs\lw.comm-server is simply named .env , no name , no other extension. 
[image]

cprezzi, 2021-06-28:
Ok, then it has to be a missing or wrong "welcome string" from the firmware. The backend (lw.comm-server) only accepts known strings to detect the firmware and version. 

Do you run the Windows EXE or the lw.comm-server from the source code? 

In case of the lw.comm-server, you could directly edit the config.js file instead of the .env file. And probably you could change the accepted welcome string for GRBL ESP in server.js (around line xxx).

ithinkido, 2021-06-28:
> 
> 
> Ok, then it has to be a missing or wrong "welcome string" from the firmware. The backend (lw.comm-server) only accepts known strings to detect the firmware and version.
> 
The start message from the board is:

[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:Init Motors]
[MSG:No spindle]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.3a ['$' for help]

> Do you run the Windows EXE or the lw.comm-server from the source code?

I am running from .exe.  I can take a look at running from scource in the next days.

cprezzi, 2021-07-02:
To run from source you can follow these instructions:
https://github.com/LaserWeb/lw.comm-server/wiki/Manual-installation-(Windows)

This starts the backend with the integrated webserver. The frontend can then be opened with a browser (prefferrably chrome) on localhost:8000.

ithinkido, 2021-07-03:
After installling Visual Studio and C++ build tools , I now have lw running from scouce now and changed the conf.js file to reset on connect.  Looking through the logs, I see a confirmtion of the connection, and what looks like a call to reconect, but no reply to the M115 firmware querry. Oddly I did try testing this borad with other senders ( LaserGRBL ) and it does connect, and gives a firmware version, so I am unable to to see what else it could be. 
[logfile.txt](https://github.com/LaserWeb/LaserWeb4/files/6758285/logfile.txt)
[image]
[image]

cprezzi, 2021-07-06:
It seems that grbl-ESP32 (unlike grbl) does not respond with a welcome string after getting a ctrl-x. Are you sure that COM7 is the right port?

If you can tell me the command to get the firmware version from grbl-ESP32, then I can help you to modify the connection procedure.

ithinkido, 2021-07-07:
> 
> 
> It seems that grbl-ESP32 (unlike grbl) does not respond with a welcome string after getting a ctrl-x. Are you sure that COM7 is the right port?

If I send a 24 ( ASCII for CRTL - X) , I get this response

[image]

cprezzi, 2021-09-27:
That seems to be correct, which means RESET_ON_CONNECT should work.

cxlso, 2022-04-06:
Hi, I run into the same problem here.. Did you get it to work? 
Thank you.