LaserWeb/LaserWeb4 #628: No supported Firmware Detected

micahpe, 2021-06-29:
Hi, I am having trouble connecting a small laser engraver to LaserWeb. It will connect for around 5 seconds and then say no supported firmware detected and disconnect itself. I have read some previous threads and tried installing and extracting a zip file but it didn't seem to work for me. Any help would be much appreciated.

cprezzi, 2021-07-02:
This is a sign that your machine firmware might not be compatible.
What exact machine do you have. 
Do you now if the firmware is grbl based?
Do you see a "welcome string" when connecting to the machine?

micahpe, 2021-07-02:
It is a generic Chinese one. I usually run it with benbox. I believe it is grbl based. When I connect it says connected to machine in green. Thanks.

cprezzi, 2021-07-02:
As far as I know, the Benbox Firmware is not grbl compatible. 
I had one of the first such devices and changed the controller to an eleksmaker board with arduino nano and grbl.

micahpe, 2021-07-02:
I don’t know much about electronics, but the board does have a nano arduino on it, so maybe and hopefully it will still work.

closed 2024-05-31 by cprezzi