LaserWeb/LaserWeb4 #631: Drag and drop opens file instead of adding it

github-throwaway, 2021-12-23:
If you drag a file to the workspace it does not get added but is rather opened. There is no way to go back, you manually have to restart LaserWeb.

[image]

[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/7770914/Laserweb-Workspace.json.txt)

cprezzi, 2021-12-27:
Drag & Drop was never implemented, and because the frontend is basically a webbrowser, the default behavior to dropping is dropping is opening the file in a new window.

easytarget, 2022-01-19:
I'll see what I can do in my work too; this looks relevant:
https://stackoverflow.com/questions/6756583/prevent-browser-from-loading-a-drag-and-dropped-file?rq=1
The trick will be still allowing drag-n-drop in the files/operations tab when assembling jobs..

easytarget, 2022-01-19:
This is now in my fork, hopefully it will be merged to the main branch soon.

Works well, doesnt interfere with in-window dragging, and guides new users into what to do.
https://github.com/easytarget-org/LaserWeb4/commit/4822c19a5e5f8350025c4f79f26deaa4e93d1000

github-throwaway, 2024-04-24:
@cprezzi should we get this merged?