LaserWeb/LaserWeb4 #634: Is it compatible with EleksLaser?

lramos-prolaser, 2022-01-14:
Hi ! I'm new to laser cutters and we have an old version of a Eleks Laser A3 ([EleksMana SE](https://wiki.eleksmaker.com/doku.php?id=eleksmana_se)) from Eleksmaker. I would like to know if I can use LaserWeb4 with this kind of laser cut. 

Thanks in advance

easytarget, 2022-01-19:
Yes; but you may need to upgrade the firmware first. Their firmware is based on GRBL and LaserWeb requires at least GRBL 1.1c. 

You should see a version number sent by the firmware in the console log when you connect, I dont know what controller software you currently have, but Pronterface is a good tool for showing this sort of low level feedback.

AFIK older Elksmakers have a firmware based on grbl 0.9, but it is unclear if elksmaker have offered any upgrades over that. 
Elksmaker do not appear to document their firmware at all ajnd they have a 'Open Source' download site that only provides binary blobs and no source whatsoever, given this level of indifference it's unlikely they will be much help.

However, the controller, despite the fancy name is a very common ATmega328 based grbl-clone board, and oplenty of people have successfullyflashed the latest GRBL to it; eg here:
https://github.com/jandelgado/eleksmaker_a3

lramos-prolaser, 2022-01-20:
Thanks to your suggestions we've managed to update GRBL and install LaserWeb! Thank you very much ! :)

closed 2022-04-07 by easytarget